steveicarus/iverilog #253: Port declaration of ha_sum inherits dimensions from var/net

mickey4u, 2019-07-09:
I have a half-adder module like this:
```
module halfadder(output logic ha_sum, output ha_carry_out, input logic ha_a, input ha_b);
      assign ha_sum = ha_a ^ ha_b;
      assign ha_carry_out  = ha_a & ha_ha_b;
endmodule
```
and it's corresponding testbench as shown below
```
`include "halfadder.sv"

module tb_halfadder();
    // test vector
    logic [3:0] testvector [3:0];
    // inputs and outputs
    logic ha_a;
    logic ha_b;
    logic ha_sum;
    logic ha_carry_out;

    // design unit instantiation
    halfadder ha(.ha_sum(ha_sum), .ha_carry_out(ha_carry_out), .ha_a(ha_a), .ha_b(ha_b));

    initial begin
        $readmemb("ha_testcases", testvector, 0, 3);
        assertOutput();
    end

    task assertOutput;
        begin
            for (int i = 0; i < 4; i++) begin
                {ha_a, ha_b, expected_sum, expected_carry_out} = testvector[i];
                #5;
                if (ha_a == 1'b0 && ha_b == 1'b0 && ha_sum == 1'b0 && ha_carry_out == 1'b0) $display("1");
                else if (ha_a == 1'b0 && ha_b == 1'b1 && ha_sum == 1'b1 && ha_carry_out == 1'b0) $display("1");
                else if (ha_a == 1'b1 && ha_b == 1'b0 && ha_sum == 1'b1 && ha_carry_out == 1'b0) $display("1");
                else if (ha_a == 1'b1 && ha_b == 1'b1 && ha_sum == 1'b0 && ha_carry_out == 1'b1) $display("1");
                else $display("0");
            end
        end
    endtask

endmodule
```

But I am getting the below errors.
```
./halfadder.sv:1: warning: Port declaration of ha_sum inherits dimensions from var/net.
sh: 1: vpphalfadder.out: not found
./halfadder.sv:1: warning: Port declaration of ha_sum inherits dimensions from var/net.
/home/michael/CLionProjects/GeLayeredFullAdder/tb_halfadder.sv:10: vvp.tgt error: uwire "ha_carry_out" must have a single driver, found (2).
error: Code generation had 1 error(s).
./halfadder.sv:1: warning: Port declaration of ha_sum inherits dimensions from var/net.
sh: 1: vpphalfadder.out: not found
./halfadder.sv:1: warning: Port declaration of ha_sum inherits dimensions from var/net
```

What could be wrong?

caryr, 2019-07-10:
The code you posted does not appear to match what generated the messages. Compiling what you posted shows that in the half added code you should be using ha_b instead of ha_ha_b and expected_sum and expected_carry_out are not defined as well as the test case vectors are missing. For more debug a complete failing example is required.

Tested with the latest V10 and devel.

mickey4u, 2019-07-10:
@caryr Below is the current codes and the error I am getting.
```
module halfadder(output logic ha_sum, ha_carry_out, input logic ha_a, ha_b);
assign ha_sum = ha_a ^ ha_b;
assign ha_carry_out = ha_a & ha_b;
endmodule
```
```
`include "halfadder.sv"

module tb_halfadder();
    // test vector
    logic [3:0] testvector [3:0];
    // inputs and outputs
    logic ha_a;
    logic ha_b;
    logic ha_sum;
    logic ha_carry_out;
    logic expected_sum;
    logic expected_carry_out;

    int passedFile;

    // design unit instantiation
    halfadder ha(.ha_sum(ha_sum), .ha_carry_out(ha_carry_out), .ha_a(ha_a), .ha_b(ha_b));

    initial begin
        $readmemb("ha_testcases", testvector, 0, 3);
        passedFile = $fopen("passed.txt", "w");
        if (!passedFile) $display("Cannot open the file");
        assertOutput();
        $fclose(passedFile);
    end

    task assertOutput;
        begin
            for (int i = 0; i < 4; i++) begin
                {ha_a, ha_b, expected_sum, expected_carry_out} = testvector[i];
                #5;
                if (ha_a == 1'b0 && ha_b == 1'b0 && ha_sum == expected_sum && ha_carry_out == expected_carry_out) $fdisplay(passedFile, "%d", 1);
                else if (ha_a == 1'b0 && ha_b == 1'b1 && ha_sum == expected_sum && ha_carry_out == expected_carry_out) $fdisplay(passedFile, "%d", 1);
                else if (ha_a == 1'b1 && ha_b == 1'b0 && ha_sum == expected_sum && ha_carry_out == expected_carry_out) $fdisplay(passedFile, "%d", 1);
                else if (ha_a == 1'b1 && ha_b == 1'b1 && ha_sum == expected_sum && ha_carry_out == expected_carry_out) $fdisplay(passedFile, "%d", 1);
                else $fdisplay(passedFile, "%d", 0);
            end
        end
    endtask

endmodule
```
```
./halfadder.sv:1: warning: Port declaration of ha_carry_out inherits dimensions from var/net.
./halfadder.sv:1: warning: Port declaration of ha_sum inherits dimensions from var/net.
```

caryr, 2019-07-10:
This compiles cleanly for me using the latest development and V10. Which version of Icarus are you using and how are you running this? Below is how I ran this using the latest devel.

> iverilog -g2009 tb_halfadder.sv
> vvp a.out
ERROR: tb_halfadder.sv:20: $readmemb: Unable to open ha_testcases for reading.

mickey4u, 2019-07-10:
@caryr ```iverilog -Wall -g2012 -o halfadder.out tb_halfadder.sv && vvp halfadder.out```

mickey4u, 2019-07-10:
@caryr  I think if you add the -Wall flag you should see it.

caryr, 2019-07-11:
Have you run this exact code because I do not get any warning messages even when -Wall is added. Also I understand wanting your code warning clean, but it should be running correctly and not have any errors like your stated originally. Is this correct?

What version of Icarus are you using?

mickey4u, 2019-07-11:
@caryr version 10.1

mickey4u, 2019-07-14:
@caryr 10.1 is the latest version or?

caryr, 2019-07-17:
10.2 is the latest, but I only have the git head for 10 and development branches compiled on my development machine.

martinwhitaker, 2019-07-23:
This was fixed sometime between v10.1 and v10.2. Please do test with the head of the v10-branch branch before reporting issues.

closed 2019-07-23 by martinwhitaker