steveicarus/iverilog #254: Why are files seemingly globbed together?

THofstee, 2019-07-22:
If I have two files, which are definitely invalid verilog files:

a.v:
```
module a(
```

b.v:
```
input logic d,
output logic q
            );

assign q = d;

endmodule
```

Doing the following returns no errors:
```
$ iverilog -g2012 a.v b.v
```

But either of the two individually fail:
```
$ iverilog -g2012 a.v 
a.v:2: syntax error
I give up.
$ iverilog -g2012 b.v 
b.v:1: syntax error
I give up.
```

If I try to pass in the `-u` flag then instead I get an error that `default_nettype directives must appear outside module definitions.`

The reason I bring this up is because I have a file that iverilog won't parse, and running a similar command spat out tons of errors about invalid syntax on a file I knew parsed properly. I would have expected `-u` to be enabled by default with `-g2012`, but in any case it still resets the compiler directives to a point where they're no longer useful because things like the default nettype no longer exist. 

Is there a way to get the behavior of `-u` while preserving things like the default nettype? Can I parse a list of verilog files and not have the errors in an individual file corrupt the remaining ones?

martinwhitaker, 2019-07-23:
From IEEE 1800-2012:

> The exact mechanism for defining which files constitute a compilation unit is tool-specific. However,
compliant tools shall provide use models that allow both of the following cases:
a) All files on a given compilation command line make a single compilation unit (in which case the
declarations within those files are accessible following normal visibility rules throughout the entire
set of files).
b) Each file is a separate compilation unit (in which case the declarations in each compilation-unit
scope are accessible only within its corresponding file).

Without -u you get (a). With -u you get (b), except that

> If there is a declaration that is incomplete at the end of a file, then the compilation unit including that file
will extend through each successive file until there are no incomplete declarations at the end of the group of files.

is not supported in this case - each file must be a complete compilation unit.

The IEEE standard requires that the compiler directives are reset at the start of each compilation unit. The whole point of supporting separate compilation units is to support separate compilation and to remove the dependency on compilation order.

The most likely reason why you get lots of syntax errors on your other file is that you are using SystemVerilog features that the Icarus parser does not support.

THofstee, 2019-07-23:
I guess I didn't mention one thing: none of my files define `default_nettype` but passing `-u` as a flag suddenly makes it `default_nettype none` (for all but I believe the first file on the arguments, need to double check that). 

Each file that compiled successfully on its own starts complaining about things like nettype when passing `-u`.

martinwhitaker, 2019-07-23:
That doesn't seem to be generally the case. This example:
````
implicit1.v:
module implicit1(a, c);

input a;
output c;

buf(b, a);
buf(c, b);

endmodule

implicit2.v:
module implicit2(a, c);

input a;
output c;

buf(b, a);
buf(c, b);

endmodule
````
when compiled with
````
iverilog -u implicit1.v implicit2.v
````
compiles without error (and fails if you add a `default_nettype none). So we need an example of code that fails.

THofstee, 2019-07-23:
Actually, it seems like the previously mentioned:

> If there is a declaration that is incomplete at the end of a file, then the compilation unit including that file will extend through each successive file until there are no incomplete declarations at the end of the group of files.

isn't exactly true.

If the first file has a syntax error or some unsupported construct, then the second file is still affected by this failure and is not compiled as its own unit.

```
iverilog -u -g2012 tap_unq1.sv CW_tap.v
tap_unq1.sv:223: syntax error
tap_unq1.sv:224: Syntax in assignment statement l-value.
tap_unq1.sv:225: syntax error
tap_unq1.sv:225: Syntax in assignment statement l-value.
tap_unq1.sv:226: syntax error
tap_unq1.sv:226: Syntax in assignment statement l-value.
tap_unq1.sv:227: syntax error
tap_unq1.sv:227: Syntax in assignment statement l-value.
tap_unq1.sv:228: syntax error
tap_unq1.sv:228: Syntax in assignment statement l-value.
tap_unq1.sv:229: syntax error
tap_unq1.sv:229: Syntax in assignment statement l-value.
tap_unq1.sv:230: syntax error
tap_unq1.sv:230: Syntax in assignment statement l-value.
tap_unq1.sv:231: syntax error
tap_unq1.sv:237: Syntax in assignment statement l-value.
tap_unq1.sv:238: syntax error
tap_unq1.sv:238: error: malformed statement
tap_unq1.sv:240: syntax error
tap_unq1.sv:240: error: malformed statement
tap_unq1.sv:242: syntax error
tap_unq1.sv:242: error: malformed statement
tap_unq1.sv:246: syntax error
tap_unq1.sv:246: error: malformed statement
tap_unq1.sv:251: syntax error
tap_unq1.sv:251: error: malformed statement
tap_unq1.sv:256: syntax error
tap_unq1.sv:256: error: malformed statement
tap_unq1.sv:263: syntax error
tap_unq1.sv:263: error: malformed statement
tap_unq1.sv:271: syntax error
I give up.
CW_tap.v:1: error: `default_nettype directives must appear
CW_tap.v:1:      : outside module definitions. The containing
CW_tap.v:1:      : module tap_unq1 starts on line tap_unq1.sv:88.
CW_tap.v:1: error: `default_nettype directives must appear
CW_tap.v:1:      : outside module definitions. The containing
CW_tap.v:1:      : module tap_unq1 starts on line tap_unq1.sv:88.
```

(CW_tap.v compiles fine on its own)

THofstee, 2019-07-23:
Here is a pair of files that triggers this behavior:

test1.sv:
```
module test(
            input logic  clk,
            input logic  d,
            output logic q
            );

   always_ff @(posedge clk) begin
      unique case(d)
        0: q <= 1;
        1: q <= 0;
      endcase // unique case (d)
   end

endmodule // test
```

test2.sv:
```
module test2(
            input logic  d,
            output logic q
            );

   assign q = d;

endmodule
```

```
iverilog -u -g2012 test1.sv test2.sv
test1.sv:8: syntax error
test1.sv:9: Syntax in assignment statement l-value.
test1.sv:10: syntax error
test1.sv:10: Syntax in assignment statement l-value.
test1.sv:11: syntax error
I give up.
test2.sv:1: error: `default_nettype directives must appear
test2.sv:1:      : outside module definitions. The containing
test2.sv:1:      : module test starts on line test1.sv:1.
test2.sv:1: error: `default_nettype directives must appear
test2.sv:1:      : outside module definitions. The containing
test2.sv:1:      : module test starts on line test1.sv:1.
```

martinwhitaker, 2019-07-23:
That should now be fixed.

THofstee, 2019-07-23:
Can confirm on my side that it seems to be fixed. Thanks for the help!

closed 2019-07-23 by THofstee