steveicarus/iverilog #261: always comb with empty sensitivity list causes segfault

galenseitz, 2019-08-16:
An always @* block that has no sensitivities will generate a warning, but is otherwise benign.
```
module foo (output logic y);
   always @* begin
      y = 1'b0;
   end
endmodule

$ iverilog -Wall '-g2012' foo.sv 
foo.sv:2: warning: @* found no sensitivities so it will never trigger.
```

An always_comb block that has no sensitivities will generate a warning, and then segfault.
```
module bar (output logic y);
   always_comb begin
      y = 1'b0;
   end
endmodule

$ iverilog -Wall '-g2012' bar.sv
bar.sv:2: warning: @* found no sensitivities so it will never trigger.
sh: line 1:  4324 Done                    /usr/local/lib/ivl/ivlpp -L -Wredef-chg -F"/tmp/ivrlg27313bae2" -f"/tmp/ivrlg7313bae2" -p"/tmp/ivrli7313bae2"
      4325 Segmentation fault      (core dumped) | /usr/local/lib/ivl/ivl -C"/tmp/ivrlh7313bae2" -C"/usr/local/lib/ivl/vvp.conf" -- -

$ iverilog -v
Icarus Verilog version 11.0 (devel) (v10_2-78-gb7b22660)
$ cat /etc/redhat-release 
CentOS Linux release 7.6.1810 (Core) 
$ cat /proc/version
Linux version 3.10.0-957.21.3.el7.x86_64 (mockbuild@kbuilder.bsys.centos.org) (gcc version 4.8.5 20150623 (Red Hat 4.8.5-36) (GCC) ) #1 SMP Tue Jun 18 16:35:19 UTC 2019
```

As a side note, the warning message is slightly misleading, as it refers to '@*' rather than 'always_comb'.

caryr, 2019-08-17:
I believe some of this is already fixed in my working repo. I will make sure it is all fixed the next time I get some free time to work on Icarus.

caryr, 2019-09-30:
This has been fixed for always_comb and the changes have been pushed to the development/master branch. Appropriate error messages are generated for always_latch/always_ff since this would not be synthesized into latch/ff behaviour.

closed 2019-09-30 by caryr