steveicarus/iverilog #262: iverilog 10_3 fails to build on Fedora 31+ (gcc 9)

rosset, 2019-08-18:
[filiperosset@raw ~]$ cd ~/rpmbuild/my/iverilog
[filiperosset@raw iverilog]$ git diff
diff --git a/iverilog.spec b/iverilog.spec
index 622fc5d..5ec860a 100644
--- a/iverilog.spec
+++ b/iverilog.spec
@@ -1,11 +1,12 @@
 Name:        iverilog
-Version:     10_2
-Release:     6%{?dist}
+Version:     10.3
+%define uver 10_3
+Release:     1%{?dist}
 Summary:     Icarus Verilog is a verilog compiler and simulator
 License:     GPLv2
 URL:         http://iverilog.icarus.com
 #fix ugly Source0
-Source0:     iverilog-10_2.tar.gz
+Source0:     iverilog-10_3.tar.gz

 BuildRequires: autoconf
 BuildRequires: bzip2-devel
@@ -23,7 +24,7 @@ engineering formats, including simulation. It strives to be true
 to the IEEE-1364 standard.

 %prep
-%autosetup
+%autosetup -n %{name}-%{uver}

 # Clean junks from tarball
 find . -type f -name ".git" -exec rm '{}' \;
@@ -68,6 +69,9 @@ make check

 %changelog
+* Sun Aug 18 2019 Filipe Rosset <rosset.filipe@gmail.com> - 10.3-1
+- Update to 10.3
+
 * Thu Jul 25 2019 Fedora Release Engineering <releng@fedoraproject.org> - 10_2-6
 - Rebuilt for https://fedoraproject.org/wiki/Fedora_31_Mass_Rebuild

[filiperosset@raw iverilog]$ fedpkg local

setting SOURCE_DATE_EPOCH=1566086400
Executing(%prep): /bin/sh -e /var/tmp/rpm-tmp.nr5i9r
+ umask 022
+ cd /home/filiperosset/rpmbuild/my/iverilog
+ cd /home/filiperosset/rpmbuild/my/iverilog
+ rm -rf iverilog-10_3
+ /usr/bin/gzip -dc /home/filiperosset/rpmbuild/my/iverilog/iverilog-10_3.tar.gz
+ /usr/bin/tar -xof -
+ STATUS=0
+ '[' 0 -ne 0 ']'
+ cd iverilog-10_3
+ /usr/bin/chmod -Rf a+rX,u+w,g-w,o-w .
+ find . -type f -name .git -exec rm '{}' ';'
++ find . -type d -name autom4te.cache -exec echo '{}' ';'
+ rm -rf
+ RPM_EC=0
++ jobs -p
+ exit 0
Executing(%build): /bin/sh -e /var/tmp/rpm-tmp.b9JQ4r
+ umask 022
+ cd /home/filiperosset/rpmbuild/my/iverilog
+ cd iverilog-10_3
+ chmod +x autoconf.sh
+ sh autoconf.sh
+ CFLAGS='-O2 -g -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -fexceptions -fstack-protector-strong -grecord-gcc-switches -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection'
+ export CFLAGS
+ CXXFLAGS='-O2 -g -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -fexceptions -fstack-protector-strong -grecord-gcc-switches -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection'
+ export CXXFLAGS
[... 1116 lines cut ...]
make[1]: Entering directory '/home/filiperosset/rpmbuild/my/iverilog/iverilog-10_3/vpi'
g++ -I. -I.. -D_LARGEFILE_SOURCE=1 -D_FILE_OFFSET_BITS=64  -DHAVE_CONFIG_H -DICARUS_VPI_CONST=const -fPIC -Wall -Wextra -Wshadow   -O2 -g -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -fexceptions -fstack-protector-strong -grecord-gcc-switches -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -MD -c vcd_priv2.cc -o vcd_priv2.o
mv vcd_priv2.d dep
make[1]: Leaving directory '/home/filiperosset/rpmbuild/my/iverilog/iverilog-10_3/vpi'
make[1]: Entering directory '/home/filiperosset/rpmbuild/my/iverilog/iverilog-10_3/vpi'
bison --verbose -t -p tblmod -d -o table_mod_parse.c table_mod_parse.y
make[1]: Leaving directory '/home/filiperosset/rpmbuild/my/iverilog/iverilog-10_3/vpi'
make[1]: Entering directory '/home/filiperosset/rpmbuild/my/iverilog/iverilog-10_3/vpi'
bison --verbose -t -p sdf -d -o sdf_parse.c sdf_parse.y
make[1]: Leaving directory '/home/filiperosset/rpmbuild/my/iverilog/iverilog-10_3/vpi'

RPM build errors:
Could not execute local: rpmbuild --define '_sourcedir /home/filiperosset/rpmbuild/my/iverilog' --define '_specdir /home/filiperosset/rpmbuild/my/iverilog' --define '_builddir /home/filiperosset/rpmbuild/my/iverilog' --define '_srcrpmdir /home/filiperosset/rpmbuild/my/iverilog' --define '_rpmdir /home/filiperosset/rpmbuild/my/iverilog' --define 'dist %{?distprefix}.fc32' --define 'fedora 32' --eval '%undefine rhel' --define 'fc32 1' -ba /home/filiperosset/rpmbuild/my/iverilog/iverilog.spec | tee .build-10.3-1.fc32.log

[filiperosset@raw iverilog]$ gcc -v
Using built-in specs.
COLLECT_GCC=/usr/bin/gcc
COLLECT_LTO_WRAPPER=/usr/libexec/gcc/x86_64-redhat-linux/9/lto-wrapper
OFFLOAD_TARGET_NAMES=nvptx-none
OFFLOAD_TARGET_DEFAULT=1
Target: x86_64-redhat-linux
Configured with: ../configure --enable-bootstrap --enable-languages=c,c++,fortran,objc,obj-c++,ada,go,d,lto --prefix=/usr --mandir=/usr/share/man --infodir=/usr/share/info --with-bugurl=http://bugzilla.redhat.com/bugzilla --enable-shared --enable-threads=posix --enable-checking=release --enable-multilib --with-system-zlib --enable-__cxa_atexit --disable-libunwind-exceptions --enable-gnu-unique-object --enable-linker-build-id --with-gcc-major-version-only --with-linker-hash-style=gnu --enable-plugin --enable-initfini-array --with-isl --enable-offload-targets=nvptx-none --without-cuda-driver --enable-gnu-indirect-function --enable-cet --with-tune=generic --with-arch_32=i686 --build=x86_64-redhat-linux
Thread model: posix
gcc version 9.1.1 20190605 (Red Hat 9.1.1-2) (GCC)
[filiperosset@raw iverilog]$

martinwhitaker, 2019-08-19:
Remove the V=1 from your invocation of make. That is overriding a variable used in the vpi Makefile. Blindly assuming that V is used to control verbosity is a mistake.

rosset, 2019-08-20:
great, V=1 or VERBOSE=1 is the default Fedora make flag for %make_build, I reverted to the old style and its fine now. Thank for your help!

-%make_build
+
+# use make, avoid use V=1 due https://github.com/steveicarus/iverilog/issues/262
+make %{?_smp_mflags}

closed 2019-08-20 by rosset