steveicarus/iverilog #263: vvp depends specifically on version 7 of readline; doesn't work with version 8

swedishhat, 2019-08-22:
I'm using [iverilog-git on Arch Linux](https://aur.archlinux.org/packages/iverilog-git/) When I run vvp, I get the following error: "/usr/local/bin/vvp: error while loading shared libraries: libhistory.so.7: cannot open shared object file: No such file or directory"

This is a shared library in **readline**. The readline version that is installed by default in Arch is version 8. I was able to fix my issue by explicitly installing [readline7 from the AUR](https://aur.archlinux.org/packages/readline7/), but I was wondering if it made sense to either make the dependency in iverilog for version 8 or allow either version.

martinwhitaker, 2019-08-23:
There is no dependency on a specific version of readline in the Icarus build scripts. I think your problem is with the AUR package. I see it hasn't been updated since 2017.

closed 2019-08-23 by martinwhitaker

swedishhat, 2019-08-23:
Sounds good. I'll flag the package in the AUR as out of date