steveicarus/iverilog #265: test.sv:51: internal error: I do not know how to elaborate (ivl_type_t) this expression.

chips4yu, 2019-09-17:
Hi,
I'm getting this error when compiling with icarus version 11.0

iverilog -g2012 test.sv

test.sv:51: internal error: I do not know how to elaborate (ivl_type_t) this expression. 
test.sv:51:               : Expression is: (bit8_lsb1)<<(axsize)
test.sv:51:               : Expression type: 8PEBShift
test.sv:61: internal error: I do not know how to elaborate (ivl_type_t) this expression. 
test.sv:61:               : Expression is: (ByteCount)-(unaligned_byte_shift)
test.sv:61:               : Expression type: 8PEBinary

What's the workaround ?
I want to shift the value of 1 up to 8 bits to the left (acting like a multiply).

Thanks,
David

test.sv

//`define C_AXI_DATA_WIDTH 32

module xyz #( parameter integer C_AXI_DATA_WIDTH = 32);

 typedef struct packed {
   bit [7-1:0]     addr;
   bit             excl;
   bit [8-1:0]     len ;
   bit [3-1:0]     size;
   bit [2-1:0]     burst;
//   bit [C_AXI_ID_WIDTH-1:0]  id;
   bit [32-1:0]  id;
 } t_cmd_message;

  typedef bit [7:0] byte_da_t[];

  function automatic byte_da_t ByteCount (input t_cmd_message Burst, input [9:0] Beat); // david 09/17/2019 added bit type to check compilation
//  function [7:0] ByteCount; // david 09/17/2019 added bit type to check compilation
    byte_da_t ByteCount; // david
//    input t_cmd_message Burst;         // burst vector
//    input [9:0]        Beat;           // beat number in the burst (1-256)
    reg   [6:0]        axaddr;
    reg   [2:0]        axsize;
    reg   [7:0]        axlen;
    reg   [1:0]        axburst;
    reg [7:0] bit8_lsb1 = 8'b00000001; // david hack
    integer bus_data_bytes;
    integer unaligned_byte_shift;
  begin
    axaddr  = Burst.addr;
    axsize  = Burst.size;
    axlen   = Burst.len;
    axburst = Burst.burst;

//    localparam integer C_AXI_DATA_WIDTH=32;
//    localparam C_AXI_DATA_WIDTH = 32;

    bus_data_bytes = C_AXI_DATA_WIDTH/8;

    // Number of bytes that the data needs to be shifted when
    // the address is unaligned
    unaligned_byte_shift =
      axaddr &              // Byte address
      ((bit8_lsb1 << axsize) - 1);      //   masked by the number of bytes
//      ((1 << axsize) - 1);      //   masked by the number of bytes
//      ((8'b00000001 << axsize) - 1);      //   masked by the number of bytes
                            //   in a transfer

    // The number of valid bits depends on the transfer size.
    ByteCount = (bit8_lsb1 << axsize);
//    ByteCount = (8'b00000001 << axsize);
//    ByteCount = (1<<axsize);
    // For FIXED bursts or on the first beat of an INCR burst
    // if the address is unaligned modify the number of
    // valid strobe bits
    if ((axburst == 0) || (Beat == 0))
    begin
      // The number of valid bits depends on the transfer size
      // and the offset of the unaligned address.
      ByteCount = ByteCount - unaligned_byte_shift;
    end
  end
  endfunction // ByteCount

endmodule

martinwhitaker, 2019-09-17:
Assigning a bit vector to a dynamic array requires an explicit cast. You need to write
````
ByteCount = byte_da_t`(bit8_lsb1 << axsize);
````
We need to output a better error message.

chips4yu, 2019-09-17:
Ok.
That fixed the first error.
But second error is still there.

test.sv:63: internal error: I do not know how to elaborate (ivl_type_t) this expression. 
test.sv:63:               : Expression is: (ByteCount)-(unaligned_byte_shift)
test.sv:63:               : Expression type: 8PEBinary

I tried type cast unaligned_byte_shift to byte_da_t type

 63 //      ByteCount = ByteCount - unaligned_byte_shift;
 64 //      ByteCount = integer'(ByteCount) - unaligned_byte_shift; // error
 65       ByteCount = ByteCount - byte_da_t'(unaligned_byte_shift);

but then got this error

test.sv:65: internal error: I do not know how to elaborate (ivl_type_t) this expression. 
test.sv:65:               : Expression is: (ByteCount)-(Unpacked array [...] of:
    vector of bool['sd7:'sd0]
'(unaligned_byte_shift))
test.sv:65:               : Expression type: 8PEBinary
1 error(s) during elaboration.

What's the workaround ?

martinwhitaker, 2019-09-19:
Think about it. You can't do arithmetic on a dynamic array, so you need to cast it back to a vector type. Then, as before, you can't assign the result back to the dynamic array without an explicit cast.

martinwhitaker, 2019-10-06:
I have fixed the compiler to output a more helpful error message when an explicit cast is required.

closed 2019-10-06 by martinwhitaker

chips4yu, 2019-10-07:
I'm not good at all this casting in systemverilog.
This is still new to me.

I tried casting the dynamic array to short integer but I'm still getting an error

66 // cast dynamic array to vector type
67     shortint'(ByteCount) = shortint'(ByteCount) - shortint'(unaligned_byte_shift);

test.sv:67: syntax error
test.sv:67: Syntax in assignment statement l-value.
test.sv:70: syntax error
test.sv:17: error: Syntax error defining function.
test.sv:3: assert: pform.cc:1355: failed assertion lexical_scope == cur_module

What's the correct syntax ?

Thanks

martinwhitaker, 2019-10-08:
Like in C/C++, you can't cast the LHS of an assignment. You have to convert the RHS expression to the right type:
`ByteCount = ByteCount'(shortint'(ByteCount) - shortint'(unaligned_byte_shift));`

chips4yu, 2019-10-09:
I get this error after your suggested update

ELABORATING DESIGN
test.sv:70: cast type and subject differ in signedness.
test.sv:70: assert: elab_expr.cc:2652: failed assertion 0

martinwhitaker, 2019-10-10:
Yes, the current implementation of casts to integral types is quite broken. I will look at fixing it.

martinwhitaker, 2019-10-11:
Casts to integer types should now be fixed.