steveicarus/iverilog #266: Add support for structs and top level local params

jotego, 2019-09-22:
Icarus Verilog does not support the System Verilog used in the [fx68k](https://github.com/ijor/fx68k) M68000 open source core.

It seems like these two features are missing:

1. Support top level **localparam**
2. Support for **typedef struct** statements

I would like to see these enhancements implemented so we can use iverilog for FOSS M68000-related projects.

mithro, 2019-09-22:
FYI - [The SymbiFlow project is developing a SystemVerilog compliance test suite](https://github.com/SymbiFlow/sv-tests). I've [added a issue to get the fx68k as a test](https://github.com/SymbiFlow/sv-tests/issues/126) but it would also be good to make sure we have specific test coverage for the features [used by the fx68k](https://github.com/ijor/fx68k).

@jotego We will then be looking at trying to help improve Icarus Verilog to support SystemVerilog and would love help.

jotego, 2019-09-23:
Thank you very much for looking into this. I’m not a software engineer myself and probably my time is better spent writing open source Verilog.

Nonetheless, I have spread the word about your needs on twitter and Discord. Hopefully reinforcements will come.

martinwhitaker, 2019-09-23:
By top-level localparam, do you mean in the compilation unit scope? That is already supported - see the sv_unit4a.v test in the test suite.

Packed structs are supported, but not unpacked.

jlwehle, 2019-10-02:
This may be a related localparam issue ... I believe Verilog 2009 added support for using localparam in the ports list ... using icarus to run:

```
  module ddc_sincos #(
    localparam SC_W = 18
      ) (
    output reg signed [SC_W - 1:0] s ,
    output reg signed [17:0]       c ,
    input signed [20:0]            x
    );

  endmodule

```
Like so:

`  iverilog -g2009 -o lpip lpip.v
`
gives:

```
  lpip.v:2: syntax error
  I give up.
```

However it works if I change localparam to parameter.

martinwhitaker, 2019-10-06:
@jlwehle, I don't think it's related, unless that's what @jotego meant by "top-level", but I have now added support for that SV feature in the master branch.

jotego, 2019-10-07:
Yes, by using the latest iVerilog code from github, rather than the compiled version from my distribution, the "top level" localparams are accepted now.

bjourne, 2022-06-24:
I suppose there is still no support for this syntax construct:
```
typedef struct {
    int rows;
    int cols;
} dims;
```
?

caryr, 2022-06-25:
You could convert that to a packed structure which is supported.

jotego, 2022-06-27:
I make my own projects compatible with what Icarus verilog and verilator can handle. But I'm also forced to use some projects that don't belong to me and that use other language features that are not supported by these tools. Making large design changes to these projects isn't an option.

I tried twice doing as you suggest for a project and I could not get the code to work after my changes. Even though I use a formal analyzer to check the changes. For large projects that you don't own, it's hard not to make a mistake when touching a fundamental design choice. 

The second problem is that the project owner is unlikely to take in a pull request that basically messes up their design for apparently no benefit. And I have also asked about it.

So, I sadly have to say that for some projects Icarus verilog is not an option until this feature gets implemented.