steveicarus/iverilog #270: Compile problem.

Kmanfi, 2019-10-07:
Hello, I've used iverilog few years on my MacBook Pro. Suddenly I cant't compile it anymore??  
What I did.  
```

  793  git clone git://github.com/steveicarus/iverilog.git
  794  cd iverilog/
  795  sh autoconf.sh
  796  ./configure
  797  make
  798  history

```
Got this.
```
g++ -I. -I.. -I../libmisc  -DHAVE_CONFIG_H -Wall -Wextra -Wshadow   -g -O2 -MD -c vtype_stream.cc -o vtype_stream.o
mv vtype_stream.d dep/vtype_stream.d
flex -s -olexor.cc ./lexor.lex
bison --verbose -t -d -o parse.cc parse.y
parse.y:2.9-16: syntax error, unexpected identifier, expecting string
make[1]: *** [parse.cc] Error 1
make: *** [all] Error 2

```
Then I tried this.  

`bison --verbose -t -d -o parse.cc parse.y
`
And make again.

```

make
mv parse.cc.h parse.h 2>/dev/null || mv parse.hh parse.h
touch parse.h
g++ -DHAVE_CONFIG_H -I. -Ilibmisc  -Wall -Wextra -Wshadow   -g -O2 -MD -c lexor.cc -o lexor.o
lexor.lex:249:17: error: use of undeclared identifier 'VLlval'; did you mean 'yylval'?
                yylval.text = yytext_string_filter(VLtext, VLleng);
                ^~~~~~
                yylval
lexor.lex:41:17: note: expanded from macro 'yylval'
# define yylval VLlval
                ^
parse.hh:967:16: note: 'yylval' declared here
extern YYSTYPE yylval;
               ^

```

martinwhitaker, 2019-10-07:
I'm guessing you are using a very old version of bison. You have two options:
1. Use a newer version (3.0.2 is known to work).
2. Edit the files `vhdlpp/parse.y` and `tgt-pcb/fp.y` and change the second line from
`%define api.pure`
to
`%pure-parser`

The `parse.cc` you manually generated is in the top-level directory. It needs to be generated with an additional bison command line option. Delete the one you created and allow make to recreate it.

Kmanfi, 2019-10-07:
You guessed it right.
```
bison --version
bison (GNU Bison) 2.3
```
Updated it to:
```
brew info bison
bison: stable 3.4.2 (bottled) [keg-only]

echo 'export PATH="/usr/local/opt/bison/bin:$PATH"' >> ~/.bash_profile
```
Compiled nice with only few warnings.

```
parse.cc:8517:18: warning: missing field 'text' initializer [-Wmissing-field-initializers]
  = { 1, 1, 1, 1 }

pform_dump.cc:252:56: warning: expression with side effects will be evaluated despite being used as an operand to 'typeid'
      [-Wpotentially-evaluated-expression]
      out << setw(indent) << "" << (type.get()? typeid(*type).name() : "<nil type>");

lexor.cc:1906:58: warning: unused parameter 'yyscanner' [-Wunused-parameter]
static void yy_fatal_error (yyconst char* msg , yyscan_t yyscanner)
                                                         ^
lexor.cc:2226:43: warning: unused parameter 'yyscanner' [-Wunused-parameter]
void *yyalloc (yy_size_t  size , yyscan_t yyscanner)
                                          ^
lexor.cc:2231:58: warning: unused parameter 'yyscanner' [-Wunused-parameter]
void *yyrealloc  (void * ptr, yy_size_t  size , yyscan_t yyscanner)
                                                         ^
lexor.cc:2243:36: warning: unused parameter 'yyscanner' [-Wunused-parameter]
void yyfree (void * ptr , yyscan_t yyscanner)
```

closed 2019-10-07 by Kmanfi

steveicarus, 2019-10-07:
I've just verified that the bison that is bundled with Xcode does NOT work.
What I am finding works fine is to use the Fink project version (
finkproject.org) instead. It is possible to install just the few bits of
tools that you need using the fink tool. I will make a note to this
effect in the README.txt.