steveicarus/iverilog #272: Imports

Kmanfi, 2019-10-07:
Two test cases for import.
Test1: I give up. 

iverilog -g2012 test1.sv
```
package MyStd;
    typedef logic [31:0] my_int32;
endpackage

package MyInterfaces;
    import MyStd::*;
    typedef struct packed
    {
        my_int32 data;
    } Response;
endpackage

module testmodule();
    import MyInterfaces::*;
    Response rsp;
initial begin
    rsp = 0;
end
endmodule
```
Test2: Compiles fine.

iverilog -g2012 test2.sv
```
package MyStd;
    typedef logic [31:0] my_int32;
endpackage

package MyInterfaces;
    //import MyStd::*;
    typedef struct packed
    {
        MyStd::my_int32 data;
    } Response;
endpackage

module testmodule();
    import MyInterfaces::*;
    MyInterfaces::Response rsp;
initial begin
    rsp = 0;
end
endmodule
```

martinwhitaker, 2019-10-07:
I can't reproduce that. Test1 compiles without error here, using
````
% bison --version
bison (GNU Bison) 3.3.2
% flex --version
flex 2.6.4
% gcc --version
gcc (Mageia 8.3.1-0.20190920.1.mga7) 8.3.1 20190920
````

Kmanfi, 2019-10-07:
Ok, so it's something here.
My setup now.
```
% bison --version
bison (GNU Bison) 3.4.2
% flex --version
flex 2.6.4
% gcc --version
Configured with: --prefix=/Library/Developer/CommandLineTools/usr --with-gxx-include-dir=/Library/Developer/CommandLineTools/SDKs/MacOSX10.14.sdk/usr/include/c++/4.2.1
Apple LLVM version 10.0.1 (clang-1001.0.46.4)
Target: x86_64-apple-darwin18.7.0
Thread model: posix
InstalledDir: /Library/Developer/CommandLineTools/usr/bin
```

martinwhitaker, 2019-10-08:
You are up to date with the master branch? Support for import into packages was only added last week.

Is "I give up" the only message you get?

Kmanfi, 2019-10-08:
Should be latest version. My colleague tested the same with his Linux and it works fine.
So, ball is on my corner. I try to find reason for this.
Branch master.

% git describe
s20150603-720-gf2ca63a5

f2ca63a5 is commit "Disable debug output when debug_elaborate is not set."

martinwhitaker, 2020-01-30:
Did you manage to resolve this?

caryr, 2020-05-09:
No response from the original poster so closing this.

closed 2020-05-09 by caryr