steveicarus/iverilog #274: $fatal should result in a non-zero exit code

ry, 2019-10-14:
```verilog
module tb;
  initial begin
    #1 if (1) $fatal(1);
    #1 $finish;
  end
endmodule
```
I would expect the executable generated with iverilog to have exit code 1, but it has exit code 0.

I'm using Icarus Verilog version 11.0 (devel) (s20150603-643-gb43fcccc)

steveicarus, 2019-10-14:
I have a patch for this ready to commit, but this behavior causes at
least one test in the regression test suite to fail, so I need to take
a little time to assess if this is really what we want.

ry, 2019-10-14:
It's standard in unix to indicate error with non-zero exit code and given that iverilog also does not print to stderr on `$fatal` means that automated tests require extra code to parse the output in order to utilize iverilog.

I'm not sure which is the best appropriate specifciation but at least [in this one](http://citeseerx.ist.psu.edu/viewdoc/download;jsessionid=EB3A1386DD588F22AE3A6413DBF278E1?doi=10.1.1.385.2407&rep=rep1&type=pdf) it states "$fatal is a run-time fatal, which shall terminates the simulation with an error code."

steveicarus, 2019-10-14:
Agreed, and that's the way I'm leaning (and also how I read the
standard) but since it impacts the environment in ways that may affect
others, I'm hoping to get other feedback on the matter as well.

mithro, 2019-10-14:
+1 to making this change. We have work arounds that are only needed for Icarus to deal with the current "return success on failure" behaviour.

steveicarus, 2019-10-15:
Fix pushed to git master.

closed 2019-10-15 by steveicarus

martinwhitaker, 2019-10-15:
Have you prepared a test suite fix?

steveicarus, 2019-10-15:
I have not.

The test that now fails is testing the $info, $warn, and $fatal system
tasks by checking the outputs. But now $fatal "fails" like it should.
But just calling it a RE will miss the check for the contents of the
output. So I decided to put the problem on the back burner and just
think about it.

caryr, 2019-10-16:
How about removing fatal from the existing test and make them separate tests. Then use RE and possibly compare to a gold files for the new tests?