steveicarus/iverilog #278: -F support

jotego, 2019-11-02:
Although iVerilog supports the -f command, it does not support another popular command: -F. The -F command processes the list of files in the indicated files relative to its folder. For instance:

-F /mylib/list.f

where list.f is
a.v
b.v
c.v

Will cause iVerilog to look for the files:
/mylib/a.v
/mylib/b.v
/mylib/c.v

rather than to look for them in the current directory. -F should be supported both at the command line and in command files.

Thank you for considering this request.

steveicarus, 2019-11-04:
Interesting.

I kinda like it, in fact. And it shouldn't be super hard to implement.

steveicarus, 2019-11-04:
I see some potential problems with this feature. Consider:

/mylib/list.f is:

$(VAR)/a.v
b.v
/otherlib/c.v

Consider also that VAR=/mylib from the environment. What happens
with -F /mylib/list.f? Suddenly, the behavior of this feature is not so
obvious. Confusing, even.

This is interesting enough that further discussion should be moved to
the iverilog-devel mailing list.

jotego, 2019-11-04:
Let me just say that the option is present in NC Verilog and Verilator.
Maybe it has some restrictions.

steveicarus, 2019-11-05:
If you can figure out how it handles being mixed with variable substitution, that would be helpful. I know that I handle the problem my using variable substitutions.

jotego, 2019-11-12:
Sorry for the delay in response. I was travelling last week and couldn't test it.
From my tests, I think the behaviour in other simulators is as follows:

_If variable substitution results in a folder path starting from root /, then do not add the local path to it. Otherwise do it._

In your example:

$(VAR)/a.v
b.v
/otherlib/c.v
where VAR = /mylib, will result in:

/mylib/a.v
/local_path/b.v
/otherlib/c.v

Local path is added only when the path to the file was already relative. Absolute paths are preserved.

jotego, 2020-07-04:
What is the status of this feature?

jotego, 2021-12-16:
Variable expansion is supported now. I think we are only missing the relative path part of **-F**

cousteaulecommandant, 2024-07-05:
Hello, let me just chime in to say that I second that relative -F inclusion would be useful :)

Currently, `-F` is supported at least by Verilator and VCS (although the documentation for the latter suggests that nested -F are not supported, but Verilator definitely supports it).

Having this can be extremely useful because it allows **modular** project design:
Say I have a module called `mymodule` in one folder of my project, and this module contains a `files.f` file like:
```
mymodule.v
adder.v
mux.v
fsm.v
```
Now, imagine I create a system on a different folder of my project which uses `mymodule`.  Its `files.f` file will have to look like:
```
mysystem.v
memory.v
bus.v
../mymodule/mymodule.v
../mymodule/adder.v
../mymodule/mux.v
../mymodule/fsm.v
```
And whenever I change something inside `mymodule/files.f` (e.g. because I added new files), I have to copy the change over to `mysystem/files.f`, prepending `../mymodule/` to every newly added line.

Instead, it would be much easier for me to just have `mysystem/files.f` be like:
```
mysystem.v
memory.v
system_bus.v
-F ../mymodule/files.f
```
so whenever `mymodule/files.f` changes, the change is reflected inside `mysystem/files.f`, while having `mymodule` and `mysystem` in two separate folders (which is cool e.g. if they are git submodules, or are maintained by different people).

Note that this cannot currently be done with plain `-f` since adding `-f ../mymodule/files.f` would be understood by iverilog, but it would look for the .v files listed there inside `mysystem/` instead of `mymodule/`, and ultimately complain that `mysystem/mymodule.v` et al do not exist.

I don't know if there's currently any solution besides writing a script that parses .f files recursively and generates a .f file including all the files.