steveicarus/iverilog #279: Timing inconsistencies?

pphilippos, 2019-11-05:
I think the following code produces incorrect results (although I am not a Verilog expert).
There seems to be a delay in one of the signals. 
Although "data" and "mode" should appear to change at the same time, they don't. 

```
module Observer(clk, mode, data);
	input clk, reset;
	input [1:0] mode;
	input [31:0] data;
		
	always @(posedge clk) $display ("Mode: %d Data: %h", mode, data);			
endmodule // Observer

module Modules(clk, valid, data);
	input clk, valid;		
	input[31:0]  data;
	
	wire [1:0] mode;	
	assign mode=(valid)?2:3;
	
	Observer o0(clk, mode, data);	
endmodule //Modules

module Top_Level();
	reg clk, valid;	
	reg [31:0] data;
	
	Modules m(clk, valid, data);	
	
	integer k;
	initial begin	
		k=1;
		valid=0;
		
		repeat(10) begin			
			valid=1; data=k;
			clk=1; #10 clk=0; #10; 	
			k=k+1;	
		end
		valid=0; data=k+1;
		clk=1; #10 clk=0; #10; 	
	end
endmodule //Top_Level
```

Output:
```
Mode: z Data: 00000001
Mode: 2 Data: 00000002
Mode: 2 Data: 00000003
Mode: 2 Data: 00000004
Mode: 2 Data: 00000005
Mode: 2 Data: 00000006
Mode: 2 Data: 00000007
Mode: 2 Data: 00000008
Mode: 2 Data: 00000009
Mode: 2 Data: 0000000a
Mode: 2 Data: 0000000c

```

Expected result:
```
Mode: 2 Data: 00000001
Mode: 2 Data: 00000002
Mode: 2 Data: 00000003
Mode: 2 Data: 00000004
Mode: 2 Data: 00000005
Mode: 2 Data: 00000006
Mode: 2 Data: 00000007
Mode: 2 Data: 00000008
Mode: 2 Data: 00000009
Mode: 2 Data: 0000000a
Mode: 3 Data: 0000000c
```

martinwhitaker, 2019-11-05:
There is a race in your code. You need to learn about Verilog scheduling semantics and non-determinism. Section 11 of IEEE 1364-2005 is the definitive guide, but try reading http://www.informit.com/articles/article.aspx?p=392278&seqNum=6 first.

pphilippos, 2019-11-05:
Oops that's why I was getting different values with Vivado. Thanks for the information

closed 2019-11-05 by pphilippos