steveicarus/iverilog #280: conditional operator with enum's require explicit cast

ryanleebunch, 2019-11-05:
The following code does not compile in iverilog.  I believe this is valid systemVerilog code.  At least one commercial tool accepts this happily.  iverilog seems to be unhappy about the datatype and wants a cast.

-------------------------------------
module enumtestcase();

typedef enum {
   		STATE_A,
   		STATE_B
	     } t_STATE ;
			 
wire mywire;  //not set just defined to illustrate error

t_STATE state;
t_STATE nxt_state;

always_comb
  case(state)
    STATE_A  : nxt_state = mywire ? STATE_B : STATE_A;  //this line is invalid in iverilog
  endcase

endmodule
----------------------------

I compiled with:
iverilog -g2012 enumtestcase.v

The error was:

enumtestcase.v:17: error: This assignment requires an explicit cast.

steveicarus, 2019-11-05:
Consider, for example, your exact example where mywire==1'bz. In that case, the result is (1'bz?1:0) which is not an enumeration of t_STATE. The expression is promoted to a bit vector then evaluated bit-wise, then assigned to the l-value. I believe this Paragraph of the IEEE 1800-2017 LRM covers the situation:

_**6.19.3 Type checking**
[...]
Enumerated variables are type-checked in assignments, arguments, and relational operators. Enumerated variables are auto-cast into integral values, but assignment of arbitrary expressions to an enumerated variable requires an explicit cast._

martinwhitaker, 2019-11-09:
I agree with Steve in the case of your example. Even if you had driven `mywire`, the compiler can't know that it will always have an unambiguous value. However, if you had used a 2-state variable or expression as the condition, I think the assignment should be allowed. I have pushed a fix for that to the master branch.

ryanleebunch, 2019-11-10:
Great!  So now the original code is valid if mywire is declared as type
bit, or if Instead I write mywire===1'b1 as the condition

martinwhitaker, 2019-11-16:
The === operator returns a 4-state value (because standard Verilog does not support 2-state values), so you will need to declare mywire as type bit.

martinwhitaker, 2019-12-23:
I think we have done as much as we can on this issue, so closing.

closed 2019-12-23 by martinwhitaker

nachiket, 2024-03-17:
I'm running into this, I'm unclear on what to case to get this to work?

iammituraj, 2024-08-02:
This passes in all industry standard simulators : VCS, Xcelium, QuestaSim without even warnings. I think it should not be an error. As the RTL behavior in an actual hardware breaks down to "if my_wire is 1'b1, assign the nxt_state to STATE_B, else assign the nxt_state to STATE_A".
So the state which is getting assigned to the nxt_state is still an enum variable, so which entity needs an explicit type cast here? The expression inside the condition should not be checked for types. 

For eg: if (my_wire) nxt_state = STATE_B; else nxt_state = STATE_A; 
Here the condition my_wire need not be enum type of state, and it should pass the compilation. It does in iverilog as well.
So, it should be the same with the ternary operator.

I think what LRM suggests is you can't assign enum with another type and it requires explicit cast. For eg : nxt_state = 1'b0 ; // This causes warning/error in the standard simulators

caryr, 2024-08-02:
Just because other tools do not complain does not negate the fact that not having a cast is an error! If you do not consider this an issue then add the cast and be done with it, but please read on for my perspective.

Read section 11.4.11 in 1800-2023 which clearly defines the conditional operator to have different functionality than an if/else statement for the case of an undefined condition. Like has been said previously, using a bit variable or possibly the case equality/inequality which are guaranteed to never return an undefined value can the cast be safely ignored. My guess is all these other simulators will mix the enumerations for an undefined condition which most likely generates an invalid enumeration value. Introducing an invalid value into an enumeration without an error or warning violates the type safety of the enumeration.

And there are actual high-Z signals in real hardware (tri-state) alone with undefined values (registers not initialized when powered on, metastability, etc.).

iammituraj, 2024-08-02:
Tristate signals are always tied to 1 or 0 in the hardware. High-Z is floating input in hardware, which is considered as noise. In simulation, High Z are tested by tieing to 1 and 0, because that's what's done in hardware as well.

iammituraj, 2024-08-02:
Tristate signals are always tied to 1 or 0 in the hardware. High-Z is floating input in hardware, which is considered as noise. In simulation, High Z are tested by tieing to 1 and 0, because that's what's done in hardware as well.

Well my point is quite simple. If "if-else" statement cannot throw error in this case, ternary operator should not throw error either. Why? Because both synthesise to the exact model in netlist.  So the tool should either comply with both or not. Either way, this should not be error, should be warning and be given option to override. As I just compiled a design with lot of FSMs, which seem to throw loads of errors in iverilog. And this is the only type of error.

iammituraj, 2024-08-02:
If an LRM feature is conflicting or may have multiple interpretations when compared with other competitive tools, the user expect to have the option to override it. Cz the user cannot change the RTL just for the compatibility across multiple tools. You can see multiple conflicts in LRM like this across tools. Usually the user will have the option to use a flag to override it.

caryr, 2024-08-02:
Did you actually read the LRM? It is very clear that the conditional operator behaves differently than an if/else statement even though they may synthesize to the same logic. The standard specifies what we are required to implement and deviating from the standard would be considered a bug. We don't implement what we want, what you want or what you expect. We implement what the standard requires and then add message, like this, to let you know when things may differ from your expectations. This difference in functionality has has been in the language for over 25 years so it's something that is unlikely to ever change.

I believe that Icarus no longer complains when the condition is a two-state variable so if you want to avoid this make the X/Z mapping to zero explicit for the conditional operator when working with enumerations. For example:

bit enum_cond;
assign enum_cond = expr_that_could_be_x_or_z; // explicitly cast x and z to false
...
STATE_A : nxt_state = enum_cond ? STATE_B : STATE_A;

I would consider the fact that other simulators do not generate a message a bug. They are assigning a likely invalid value to an enumeration instead of flagging that as an error when the condition is unknown. You could argue that this should be a runtime check and only trigger if an invalid value is actually assigned or that the message should be changed to provide more details, but this is certainly a case where an invalid value could be assigned and we choose to report this as a compilation error. The above code makes the users intention explicit and should give the same result for any simulator.

If we had a functioning $cast() system function then it could be used to dynamically check the assignment value is valid, but runtime checks have runtime overhead and we prefer static checks where they make sense.

iammituraj, 2024-08-03:
Using "bit" is not at all a work around. It masks x-propagation and undriven nets in the design, leading to bugs. Which are uncovered only in gate level simulation later. So, that's highly unrecommended.

SV has always been historically less strictly typed language unlike VHDL. In this case, there is no explicit type casting required at all because type casting should be done on the final variable which is getting assigned to left side. Here both are enum state variables, and they are already same data type. But the tool is trying to enforce type casting on the condition inside ternary operator rather, which doesn't make any sense at all. It's like enforcing typecasting on the condition within the if statement, which is not a requirement at all in LRM. Probably, this is why other tools don't complain???

My interpretation of the LRM statement on type casting of enum is - explicit typecasting should be done on the RHS variable which is assigned to the enum state variable on LHS. Yes, otherwise it's error in all compilers I used. But an expression inside condition is not assigning anything to LHS,  therefore no need to type cast that ...As I said this could be open for multiple interpretations as there is no example in LRM. I expect this to be not error. If iverilog has different interpretation on this from other tools, it should be warning atmost, not error. Cz no user wants to explicitly change their core RTL codes for comparability across tools. And for static checks, we use linter rather, not compiler.

You argue that x or z inside ternary operator condition can lead to undefined state assignment. That's true. But that's true for any other construct in SV. It would propagate z or x for "if" statement as well the condition was x or z.  Same for assign statements. X propagation is problem as this could be a bug in the design. That's why verification is done, there is it is caught anyway. 

You agreed to me that both will synthesise to same logic. That's precisely why simulator should have shown same behavior for both 'if' and ternary operator in this particular case. A simulator is supposed to behave like what hardware behaves. Otherwise I don't know what is a simulator's job.

iammituraj, 2024-08-03:
Therefore, I would recommend the developers of iverilog to take another look into this case. As a user, I found this rather unagreeable as "ERROR". Especially when it compiles clean on every other industry compilers I tried.

caryr, 2024-08-03:
You clearly have not read the part of the standard I suggested. It describes why the conditional operator is mandated to simulate differently than an if/else statement even though they may synthesize into the same gates. What I find very interesting is my use of bit, which you criticize as leading to bugs, converts the conditional operator to simulate exactly the same as a standard compliant if/else which you seem to have no problem with. I'm trying to educate you why the conditional operator creates type safety issues in SystemVerilog and instead, you want to argue that since other tools support it Icarus should as well. Maybe Icarus is the odd one out, but that does not imply what it is doing is wrong. If you believe Icarus is wrong then you need to find where in the standard your expected behavior is described as valid otherwise we don't have too much to discuss. Maybe it's time for me to work on getting the other tools in line because this is clearly a type safety issue for enumeration assignments.

`It would propagate z or x for "if" statement as well if the condition was x or z.` this is only true if non-standard x-propagation is enabled and while non-standard behavior is allowed when requested it is not how a standard compliant simulator should behave. By default an X/Z is considered false.

caryr, 2024-08-03:
This is from section 6.19.3 of 1800-2023:

> Enumerated types are strongly typed; thus, a variable of type enum cannot be directly assigned a value that
> lies outside the enumeration set unless an explicit cast is used or unless the enum variable is a member of a
> union.

Which implies violating type safety should be a compilation error.