steveicarus/iverilog #281: SystemVerilog void function issue

apseng03, 2019-11-07:
Hello,
  I am trying to compile a simple void function in latest (devel) iverilog, Without "void" it works, albeit with a warning (valid one). However, with "void", I get an error. Please see below for code and commands:

Working: iverilog -g2012 void_fn.sv 
Not working: iverilog -g2012 -DBUG void_fn.sv 

Please look into the issue.

Thanks
Ani

// code void_fn.sv
`ifndef VW_IVL_GO2UVM_PKG
  `define VW_IVL_GO2UVM_PKG

package vw_ivl_go2uvm_pkg;
  int uvm_err_counter;
  int uvm_warn_counter;
  int uvm_info_counter;

  `ifndef BUG
  function uvm_count_warn();
    uvm_warn_counter++;
  endfunction : uvm_count_warn

  function uvm_count_err();
    uvm_err_counter++;
  endfunction : uvm_count_err

  function uvm_count_info();
    uvm_info_counter++;
  endfunction : uvm_count_info

`else 
  function void uvm_count_err();
    uvm_err_counter++;
  endfunction : uvm_count_err

  function void uvm_count_warn();
    uvm_warn_counter++;
  endfunction : uvm_count_warn

  function void uvm_count_info();
    uvm_info_counter++;
  endfunction : uvm_count_info
 `endif // BUG

endpackage : vw_ivl_go2uvm_pkg

`endif //  VW_IVL_GO2UVM_PKG

module top;
  import vw_ivl_go2uvm_pkg::*;

  initial begin
    uvm_count_info();
  end

endmodule : top

iverilog -g2012 void_fn.sv 
void_fn.sv:48: warning: User function 'uvm_count_info' is being called as a task.
[Anirudh-MacBook-Air:IVLOG_SV_VW/VW_IVLOG_GO2UVM/run_dir] ani% vvp a.out 
[Anirudh-MacBook-Air:IVLOG_SV_VW/VW_IVLOG_GO2UVM/run_dir] ani% iverilog -g2012 -DBUG void_fn.sv
void_fn.sv:48: warning: User function 'uvm_count_info' is being called as a task.
void_fn.sv:48: assert: elab_expr.cc:1272: failed assertion 0
sh: line 1: 13077 Done                    /Users/ani/tools/ivl_devel/lib/ivl/ivlpp -L -F"/var/folders/rk/7dbwr18d6f9g7w9yqb06s6yh0000gn/T//ivrlg2d192465" -f"/var/folders/rk/7dbwr18d6f9g7w9yqb06s6yh0000gn/T//ivrlgd192465" -p"/var/folders/rk/7dbwr18d6f9g7w9yqb06s6yh0000gn/T//ivrlid192465"
     13078 Abort trap: 6           | /Users/ani/tools/ivl_devel/lib/ivl/ivl -C"/var/folders/rk/7dbwr18d6f9g7w9yqb06s6yh0000gn/T//ivrlhd192465" -C"/Users/ani/tools/ivl_devel/lib/ivl/vvp.conf" -- -
[Anirudh-MacBook-Air:IVLOG_SV_VW/VW_IVLOG_GO2UVM/run_dir] ani%

steveicarus, 2019-11-07:
Pushed a fix to git master.

closed 2019-11-07 by steveicarus

apseng03, 2019-11-09:
Hello,
Thanks for the earlier fix. Now it shows a different error. The error is that there are too many arguments. 

**Command**: iverilog -g2012 -DBUG void_fn.sv
**Error**: ../ivl_uvm_tests/void_fn.sv:45: error: Too many arguments (1, expecting 0) in call to task.
1 error(s) during elaboration.

Thanks,
Ani

martinwhitaker, 2019-11-09:
Should be fixed now in git master.

apseng03, 2019-11-10:
Hello,
Thanks for the fix.
Thanks,
Ani