steveicarus/iverilog #283: Expression evaluating to 1'bx instead of expected 1'b0

ymherklotz, 2019-11-12:
The following code outputs and assigns `y` to 1'bx instead of 1'b0. This happens in iverilog version 10.3 and also happens in version 10.0 (11/23/14) on edaplayground. However, this seems to execute fine in version 9.6 and 9.7 on edaplayground.

I have a link to the design loaded in [edaplayground](https://www.edaplayground.com/x/5SMW).

```verilog
module testbench;
   wire y;
   reg  clk;
   reg [15:0] wire0; // Reducing the size of wire0 to 15 bits seems to fix the output.
   reg        reg1 = 1'h0;
   assign y = reg1;
   initial
     begin
        clk = 1'h0;
        wire0 = 1'h0;
        #10 $finish;
     end
   always #5 clk = ~clk;
   always @(posedge clk) $strobe("%b", y);
   // Problematic line, should assign 1'b0 but assigns 1'bx.
   always @(posedge clk) reg1 <= (wire0 >> {wire0 ~^ 1'b1, wire0});
endmodule
```

After running

```
iverilog testbench.v -o im && ./im
```

the output is `x` instead of `0`.

If the always block is not used, and the expression is assigned to `y` directly, the result is the expected 1'b0. Any changes to the expression seems to fix the result as well.

martinwhitaker, 2019-11-13:
I assume this is an artificial stress test, as the expression should evaluate to zero for any unambiguous value of wire0.

The problem is that the shift distance evaluates to a large 32-bit value (32'hfffe0000). The vvp %shiftr instruction casts the value to an int before performing the shift operation, thus converting it to a left shift.

ymherklotz, 2019-11-13:
Ah OK thanks for explaining the reasoning for that. I was testing Quartus using randomly generated designs and was surprised that an undefined value was present in the output.

martinwhitaker, 2019-11-16:
Fixed in both master and v10 branches.

closed 2019-11-16 by martinwhitaker