steveicarus/iverilog #284: Warning in large integers module attributes

cassiersg, 2019-11-15:
For the following module:
```
(* x= 10000000000000000000 *)
module a();
endmodule
```
iverilog produces the warning
```
warning: verinum::as_long() truncated 65 bits to 63, returns 776627963145224192
```

This seems to have no other impact on my simulation (I am not using this attribute elsewhere, it is used by other tools). I would therefore not be annoyed by a WONTFIX answer.

Interstingly, the following snippet doesn't produce any warning:
```
module a();
(* x= 1000000000000000000000 *) wire x;
endmodule
```

iverilog version: git 4a8f4d65d0a3

martinwhitaker, 2019-11-17:
The warning is produced when the attribute is passed to the target code generator. The ivl_target API passes numeric attributes in a native "long int".

There is no warning for your second code snippet because when using the default vvp target, unused signals are removed from the design before code generation.

We could discard numeric attributes that are too large to pass to the code generator, either with a warning or silently. I'll wait to see if Steve has an opinion on this.

steveicarus, 2019-11-17:
Yikes, that's an interesting problem. It might go deeper than that.
The standard doesn't seem to make clear if numeric values are bit or
logic, so (* x = 'b1x00 *) may be handled incorrectly at the
ivl_target.h API.

I think this should be taken to the iverilog-devel mailing list and
hashed out there.