steveicarus/iverilog #285: I can't install iverilog on Ubuntu 18.04

lzclizechen, 2019-11-18:
Hello！When I use “make check” after “sh autoconf.sh” and " ./configure ", it report like this : 
----------------------------
bison --verbose -t -p VL -d -o parse.cc parse.y
parse.y:459 parser name defined to default :"parse"
parse.y:1123:  warning:  type clash ('' 'expr') on default action
parse.y:1124:  warning:  type clash ('' 'expr') on default action
parse.y:1125:  warning:  type clash ('' 'expr') on default action
parse.y:1158:  warning:  type clash ('' 'named_pexprs') on default action
parse.y:1786:  warning:  type clash ('' 'property_qualifier') on default action
parse.y:2029:  warning:  type clash ('' 'expr') on default action
parse.y:2221:  warning:  type clash ('' 'expr') on default action
parse.y:3371:  warning:  type clash ('' 'event_statement') on default action
parse.y:5231:  warning:  type clash ('' 'named_pexprs') on default action
mv parse.cc.h parse.h 2>/dev/null || mv parse.hh parse.h
mv: cannot stat 'parse.hh': No such file or directory
Makefile:261: recipe for target 'parse.h' failed
make: *** [parse.h] Error 1
----------------------------
I'm crasy now, help me please!

steveicarus, 2019-11-18:
Looks like an issue with the version of bison? What's the output from "bison --version"?

lzclizechen, 2019-11-18:
Thanks for reply. the version is 3.0.4

lzc@ubuntu:~$ bison --version
bison (GNU Bison) 3.0.4

lzclizechen, 2019-11-18:
Hello , I made a fake parse.hh to the dir iverilog (copy from parse.h) and it seemed worked . But a new problem appeared:

flex -s -olexor.cc ./lexor.lex
"./lexor.lex", line 4: unrecognized %option: reentrant
Makefile:116: recipe for target 'lexor.cc' failed
make[1]: *** [lexor.cc] Error 1
make[1]: Leaving directory '/home/lzc/iverilog/vhdlpp'
Makefile:126: recipe for target 'all' failed
make: *** [all] Error 2

martinwhitaker, 2019-11-19:
I have just installed Ubuntu 18.04 in a VM and successfully built from a fresh clone of git master. So there are no problems with the standard tool versions in 18.04.

You don't mention what source you are building from. Is it possible a file has got corrupted?

caryr, 2019-11-20:
I regularly build the latest development using 18.04 in a VM and just like Martin have not had any issues. Which shell is make using? In my Makefile it is /bin/sh and flex is version 2.6.4

martinwhitaker, 2019-12-23:
@lzclizechen, have you resolved this? Without more information, there's not much we can do.

martinwhitaker, 2020-01-30:
No response so closing as works-for-me. Feel free to reopen if you can provide more information.

closed 2020-01-30 by martinwhitaker