steveicarus/iverilog #287: buffer overflow detected

mithro, 2019-11-21:
Steps to reproduce;
 1. Compile iverilog from commit `s20150603-770-g351a4e5f` with libfortify.
 2. Try to run iverilog on the following file;
    ```verilog
    module main;
      initial
        begin
          $display("Hello, World");
          $finish ;
        end
    endmodule
    ```
 3. Get a segfault.

Looks like this started happening between `s20150603_0738_g16e8563c` `s20150603_0761_g7feb26ff`?

See log output below;

```
Icarus Verilog version 11.0 (devel) (s20150603-770-g351a4e5f-dirty)
Copyright 1998-2019 Stephen Williams
  This program is free software; you can redistribute it and/or modify
  it under the terms of the GNU General Public License as published by
  the Free Software Foundation; either version 2 of the License, or
  (at your option) any later version.
  This program is distributed in the hope that it will be useful,
  but WITHOUT ANY WARRANTY; without even the implied warranty of
  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
  GNU General Public License for more details.
  You should have received a copy of the GNU General Public License along
  with this program; if not, write to the Free Software Foundation, Inc.,
  51 Franklin Street, Fifth Floor, Boston, MA 02110-1301, USA.
Icarus Verilog Preprocessor version 11.0 (devel) (s20150603-770-g351a4e5f-dirty)
Copyright (c) 1999-2017 Stephen Williams (steve@icarus.com)
  This program is free software; you can redistribute it and/or modify
  it under the terms of the GNU General Public License as published by
  the Free Software Foundation; either version 2 of the License, or
  (at your option) any later version.
  This program is distributed in the hope that it will be useful,
  but WITHOUT ANY WARRANTY; without even the implied warranty of
  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
  GNU General Public License for more details.
  You should have received a copy of the GNU General Public License along
  with this program; if not, write to the Free Software Foundation, Inc.,
  51 Franklin Street, Fifth Floor, Boston, MA 02110-1301, USA.
Icarus Verilog Parser/Elaborator version 11.0 (devel) (s20150603-770-g351a4e5f-dirty)
Copyright (c) 1998-2019 Stephen Williams (steve@icarus.com)
  This program is free software; you can redistribute it and/or modify
  it under the terms of the GNU General Public License as published by
  the Free Software Foundation; either version 2 of the License, or
  (at your option) any later version.
  This program is distributed in the hope that it will be useful,
  but WITHOUT ANY WARRANTY; without even the implied warranty of
  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
  GNU General Public License for more details.
  You should have received a copy of the GNU General Public License along
  with this program; if not, write to the Free Software Foundation, Inc.,
  51 Franklin Street, Fifth Floor, Boston, MA 02110-1301, USA.
 FLAGS DLL vvp.tgt
vvp.tgt: Icarus Verilog VVP Code Generator 11.0 (devel) (s20150603-770-g351a4e5f-dirty)
Copyright (c) 2001-2015 Stephen Williams (steve@icarus.com)
  This program is free software; you can redistribute it and/or modify
  it under the terms of the GNU General Public License as published by
  the Free Software Foundation; either version 2 of the License, or
  (at your option) any later version.
  This program is distributed in the hope that it will be useful,
  but WITHOUT ANY WARRANTY; without even the implied warranty of
  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
  GNU General Public License for more details.
  You should have received a copy of the GNU General Public License along
[... 99 lines cut ...]
7feda71e8000-7feda71ec000 rw-p 0016c000 08:01 2055100                    /tmp/really-really-really-really-really-really-really-really-really-really-really-really-really-long-path/conda/conda-bld/iverilog_1574373914010/_test_env_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehol/lib/libstdc++.so.6.0.26
7feda71ec000-7feda71ef000 rw-p 00000000 00:00 0 
7feda71fd000-7feda71ff000 rw-p 00000000 00:00 0 
7feda71ff000-7feda7200000 r--p 00025000 08:01 55463                      /lib/x86_64-linux-gnu/ld-2.23.so
7feda7200000-7feda7201000 rw-p 00026000 08:01 55463                      /lib/x86_64-linux-gnu/ld-2.23.so
7feda7201000-7feda7202000 rw-p 00000000 00:00 0 
7ffcdc6c1000-7ffcdc6e5000 rw-p 00000000 00:00 0                          [stack]
7ffcdc7e3000-7ffcdc7e6000 r--p 00000000 00:00 0                          [vvar]
7ffcdc7e6000-7ffcdc7e8000 r-xp 00000000 00:00 0                          [vdso]
ffffffffff600000-ffffffffff601000 r-xp 00000000 00:00 0                  [vsyscall]
Aborted (core dumped)
Icarus Verilog version 11.0 (devel) (s20150603-770-g351a4e5f-dirty)
Copyright 1998-2019 Stephen Williams
  This program is free software; you can redistribute it and/or modify
  it under the terms of the GNU General Public License as published by
  the Free Software Foundation; either version 2 of the License, or
  (at your option) any later version.
  This program is distributed in the hope that it will be useful,
  but WITHOUT ANY WARRANTY; without even the implied warranty of
  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
  GNU General Public License for more details.
  You should have received a copy of the GNU General Public License along
  with this program; if not, write to the Free Software Foundation, Inc.,
  51 Franklin Street, Fifth Floor, Boston, MA 02110-1301, USA.
```

martinwhitaker, 2019-11-22:
There is not enough information here for me to reproduce the problem. There is no "libfortify" available in my Linux distribution, and Google doesn't help me. Taking a wild stab, I recompiled with -D_FORTIFY_SOURCE=2, but that didn't reveal anything.

steveicarus, 2019-11-22:
Hmm.... I'm guessing that it has something to do with the very very long
install paths you are using. The ivl program uses the install path to find
modules when generating the .vvp output file, specifically the module
paths. I don't see any fixed size buffers in that bit of code, though.

Also, you are using a -dirty version?

mithro, 2019-11-22:
FYI - Probably not that more helpful, but the full log of the build and test run is at https://travis-ci.com/mithro/litex-conda/jobs/259295977

It looks like you should just need to do `CPPFLAGS=-DNDEBUG -D_FORTIFY_SOURCE=2 -O2` with GCC newer than 4.0 to get the fortification -- (`_FORTIFY_SOURCE` doesn't do anything without at least `-O1` apparently? - https://access.redhat.com/blogs/766093/posts/3606481). 

Just trying to repro locally and not on our CI system.

mithro, 2019-11-22:
```
LOCATING TOP-LEVEL MODULES
   main
 ... done, 0 seconds.
ELABORATING DESIGN
 ... done, 0 seconds.
RUNNING FUNCTORS
 -F cprop ...
 ... Iteration detected 0 optimizations.
 ... Look for dangling constants
 ... done
 -F nodangle ...
 ... scan for dangling signal and event nodes. (scomplete=F, ecomplete=F)
 ... 1 iterations deleted 0 dangling signals and 0 events.
 ... done
CALCULATING ISLANDS
 ... done, 0 seconds.
CODE GENERATION
 ... invoking target_design
*** buffer overflow detected ***: /usr/local/google/home/tansell/conda/conda-bld/iverilog_1574387762355/_test_env_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_pla/lib/ivl/ivl terminated

Thread 4.1 "ivl" received signal SIGABRT, Aborted.
__GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
50      ../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
─── Assembly ───────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
0x00007ffff74c9cef __GI_raise+255 mov    $0x2,%edi
0x00007ffff74c9cf4 __GI_raise+260 mov    $0xe,%eax
0x00007ffff74c9cf9 __GI_raise+265 syscall 
0x00007ffff74c9cfb __GI_raise+267 mov    0x108(%rsp),%rcx
0x00007ffff74c9d03 __GI_raise+275 xor    %fs:0x28,%rcx
0x00007ffff74c9d0c __GI_raise+284 mov    %r8d,%eax
0x00007ffff74c9d0f __GI_raise+287 jne    0x7ffff74c9d2e <__GI_raise+318>
─── Expressions ────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
─── History ────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
─── Memory ─────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
─── Registers ──────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
   rax 0x0000000000000000    rbx 0x0000000000000006    rcx 0x00007ffff74c9cfb    rdx 0x0000000000000000    rsi 0x00007fffffffaba0    rdi 0x0000000000000002    rbp 0x00007fffffffaf80    rsp 0x00007fffffffaba0     r8 0x0000000000000000     r9 0x00007fffffffaba0    r10 0x0000000000000008    r11 0x0000000000000246    r12 0x00007fffffffae40    r13 0x0000000000001000    r14 0x0000000000000040    r15 0x00007ffff7fcc000
   rip 0x00007ffff74c9cfb eflags [ PF ZF IF ]           cs 0x00000033             ss 0x0000002b             ds 0x00000000             es 0x00000000             fs 0x00000000             gs 0x00000000        
─── Source ─────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
Cannot display "/build/glibc-XAwaOT/glibc-2.28/signal/../sysdeps/unix/sysv/linux/raise.c" ([Errno 2] No such file or directory: '/build/glibc-XAwaOT/glibc-2.28/signal/../sysdeps/unix/sysv/linux/raise.c')
─── Stack ──────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
[0] from 0x00007ffff74c9cfb in __GI_raise+267 at ../sysdeps/unix/sysv/linux/raise.c:50
arg sig = 6
[1] from 0x00007ffff74b48ad in __GI_abort+297 at abort.c:79
(no arguments)
[+]
[... cut ...]
#6  0x00007ffff759a879 in __strncpy_chk (s1=<optimized out>, s2=<optimized out>, n=<optimized out>, s1len=<optimized out>) at strncpy_chk.c:26
#7  0x00007ffff7fd592a in target_design () from /usr/local/google/home/tansell/conda/conda-bld/iverilog_1574387762355/_test_env_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_pla/lib/ivl/vvp.tgt
#8  0x0000555555734e6a in dll_target::end_design(Design const*) ()
#9  0x000055555566a244 in Design::emit(target_t*) const ()
#10 0x00005555555e30e5 in main ()
```

mithro, 2019-11-22:
Still trying to get a more useful traceback.

mithro, 2019-11-22:
```
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
#1  0x00007ffff74b48ad in __GI_abort () at abort.c:79
#2  0x00007ffff750bca8 in __libc_message (action=<optimized out>, fmt=fmt@entry=0x7ffff7615d5b "*** %s ***: %s terminated\n") at ../sysdeps/posix/libc_fatal.c:181
#3  0x00007ffff759d1bd in __GI___fortify_fail_abort (need_backtrace=need_backtrace@entry=true, msg=msg@entry=0x7ffff7615cd8 "buffer overflow detected") at fortify_fail.c:28
#4  0x00007ffff759d1f1 in __GI___fortify_fail (msg=msg@entry=0x7ffff7615cd8 "buffer overflow detected") at fortify_fail.c:44
#5  0x00007ffff759b2e0 in __GI___chk_fail () at chk_fail.c:28
#6  0x00007ffff759a879 in __strncpy_chk (s1=s1@entry=0x7fffffffabb0 "\001", s2=s2@entry=0x555555826790 "/usr/local/google/home/tansell/conda/conda-bld/iverilog_1574389867081/_test_env_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_"..., n=n@entry=272, s1len=s1len@entry=128) at strncpy_chk.c:26
#7  0x00007ffff7fd592a in strncpy (__len=272, __src=0x555555826790 "/usr/local/google/home/tansell/conda/conda-bld/iverilog_1574389867081/_test_env_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_"..., __dest=0x7fffffffabb0 "\001") at /usr/local/src/conda-prefix/x86_64-conda_cos6-linux-gnu/sysroot/usr/include/bits/string3.h:121
#8  draw_module_declarations (des=0x5555557f7d00 <dll_target_obj+32>) at vvp.c:90
#9  target_design (des=0x5555557f7d00 <dll_target_obj+32>) at vvp.c:220
#10 0x0000555555734e6a in dll_target::end_design (this=0x5555557f7ce0 <dll_target_obj>) at t-dll.cc:752
#11 0x000055555566a244 in Design::emit (this=this@entry=0x555555815b20, tgt=0x5555557f7ce0 <dll_target_obj>) at emit.cc:567
#12 0x00005555555e30e5 in main (argc=<optimized out>, argv=0x7fffffffd158) at main.cc:1290
```

mithro, 2019-11-22:
```
basedir:/usr/local/google/home/tansell/conda/conda-bld/iverilog_1574387762355/_test_env_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_pla/lib/ivl
module:/usr/local/google/home/tansell/conda/conda-bld/iverilog_1574387762355/_test_env_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_pla/lib/ivl/system.vpi
module:/usr/local/google/home/tansell/conda/conda-bld/iverilog_1574387762355/_test_env_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_pla/lib/ivl/vhdl_sys.vpi
module:/usr/local/google/home/tansell/conda/conda-bld/iverilog_1574387762355/_test_env_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_pla/lib/ivl/vhdl_textio.vpi
module:/usr/local/google/home/tansell/conda/conda-bld/iverilog_1574387762355/_test_env_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_pla/lib/ivl/v2005_math.vpi
module:/usr/local/google/home/tansell/conda/conda-bld/iverilog_1574387762355/_test_env_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_pla/lib/ivl/va_math.vpi
generation:2005
generation:no-specify
generation:assertions
generation:xtypes
generation:io-range-error
generation:no-strict-ca-eval
generation:no-strict-expr-width
generation:shared-loop-index
generation:no-verilog-ams
generation:icarus-misc
warnings:n
ignore_missing_modules:false
out:hello_world
iwidth:32
widthcap:65536
ivlpp:/usr/local/google/home/tansell/conda/conda-bld/iverilog_1574387762355/_test_env_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_pla/lib/ivl/ivlpp  -L -F"/tmp/ivrlg262d32cd9" -P"/tmp/ivrli62d32cd9"
```

Issue seems to be;
https://github.com/steveicarus/iverilog/blob/f147cf9c61b7d9ce2528e92fc757f583ebccc4e3/tgt-vvp/vvp.c#L78-L96

mithro, 2019-11-22:
Seems to be because my path is 272 characters long which is >128 that buffer is.

steveicarus, 2019-11-22:
I see the problem now. I pushed a fix.

mithro, 2019-11-22:
Should this be 
```c
 __inline__ static void draw_module_declarations(ivl_design_t des) 
 { 
       const char*cp = ivl_design_flag(des, "VPI_MODULE_LIST"); 
  
       while (*cp) { 
 	    char buffer[PATH_MAX]; 
 	    const char*comma = strchr(cp, ','); 
  
 	    if (comma == 0) 
 		  comma = cp + strlen(cp); 
  
 	    strncpy(buffer, cp, comma-cp); 
 	    buffer[comma-cp] = 0; 
 	    fprintf(vvp_out, ":vpi_module \"%s\";\n", buffer); 
  
 	    cp = comma; 
 	    if (*cp) cp += 1; 
       } 
 } 
```
? https://github.com/torvalds/linux/blob/master/include/uapi/linux/limits.h#L13

(However that still has issues according to http://insanecoding.blogspot.com/2007/11/pathmax-simply-isnt.html and https://eklitzke.org/path-max-is-tricky ?)

mithro, 2019-11-22:
Looks like `PATH_MAX` is already used in [`vpi/fstapi.c`](vpi/fstapi.c) so that might be an okay solution?

steveicarus, 2019-11-22:
The buffer should be dynamically allocated (and freed). I've pushed a fix to git master and git v10-branch. Give it a try, and if it fixes the issue, we can close this report.

mithro, 2019-11-22:
Seems to be fixed! THank you for your superspeed response.

closed 2019-11-22 by steveicarus