steveicarus/iverilog #288: Syntax error when using parameter to set net width

gemarcano, 2019-11-25:
Disclaimer: I'm brand new to verilog (but am a professional software developer).

I have been tinkering on an icebreaker FPGA dev kit for a few weeks, and I've begun reading more into setting up testbenches to properly test my designs instead of throwing them at the FPGA and seeing what happens. Thus far I've been using yosys to synthesize my verilog designs. Today I downloaded iverilog 10.3 and it began complaining about `test.v:3: error: syntax error in continuous assignment`. I reduced the verilog file to the following that still reproduces the error:
```
module test(output [WIDTH-1:0] out);
    parameter WIDTH = 16;
    assign out = WIDTH'b1;
endmodule
```
If I replace the `WIDTH'b1` with `16b1` explicitly, the syntax error vanishes. Am I doing something wrong or is this a bug?

This seems to be similar to #94 (or related?).

gemarcano, 2019-11-25:
In case it wasn't clear, a (slightly) larger design with similar syntax that fails equally in iverilog synthesizes with yosys.

gemarcano, 2019-11-25:
I forgot to mention, while I did download and install 10.3, I also cloned master, compiled and installed it (to /usr/local as to not clobber anything else), then tested it. `/usr/local/bin/iverilog -v` returns: `Icarus Verilog version 11.0 (devel) (s20150603-773-gb25df08c)` (and copyright information).

martinwhitaker, 2019-11-25:
That is illegal syntax. The IEEE standard says:
````
binary_number ::= [ size ] binary_base binary_value
size ::= non_zero_unsigned_number
non_zero_unsigned_number ::= non_zero_decimal_digit { _ | decimal_digit}
````
i.e. the size must be a literal value.

gemarcano, 2019-11-25:
Unless I'm misinterpreting #94, does that mean that behavior there is also using illegal syntax?

martinwhitaker, 2019-11-25:
No. Note the brackets. That is a SystemVerilog size cast.

May I suggest you obtain a copy of the LRM. You can download it for free, thanks to the IEEE GET Program: https://ieeexplore.ieee.org/browse/standards/get-program/page/series?id=80

gemarcano, 2019-11-25:
I'm not sure I understand what brackets you're referring to, the examples I'm seeing on that issues include lines such as `dout = WIDTH'(din + 1);`, which is what I'm referring to (I don't see brackets there). It could be SystemVerilog handles that differently than Verilog 2005.

I did take a look at the Verilog 2005 specification, and I agree with your interpretation of it, I'm not sure why yosys was accepting it (maybe it has a weird hybrid SystemVerilog/Verilog parser?). Since this isn't an issue with iverilog in Verilog mode, I'll go ahead and close the issue. I think I've found the right way to do what I was trying to do in Verilog (using concatenation and replication). Thanks!

closed 2019-11-25 by gemarcano

martinwhitaker, 2019-11-26:
The line you quote has "(" and ")" brackets. '( ... ) is the syntax for a SystemVerilog cast.

gemarcano, 2019-11-28:
Ah, good to know. Thanks again! I'll be using iverilog to make sure my syntax is good, since yosys seems to be too forgiving.