steveicarus/iverilog #289: Member el2 of packed union is 4 bits, expecting 0 bits.

patrick96, 2019-12-03:
I started working in a larger systemverilog codebase and wanted to simulate it using iverilog. But I was getting an error during elaboration. I was able to track it down to the following minimal example:

```systemverilog
package pkg;

typedef enum logic [3:0] { 
    ABC = 4'h0
} enum_t;

typedef struct packed {
    enum_t item;
} w_enum;

typedef struct packed {
    logic [3:0] item;
} w_logic;

typedef union packed {
    w_enum el1;
    w_logic el2;
} foo_t;

endpackage

import pkg::*;

module main();
foo_t val;
endmodule
```

Compiling this with `iverilog -v -g2012 main.sv` gives me the following error message (I stripped the copyright notice):

```
Icarus Verilog version 11.0 (devel) (s20150603-774-g7f95abc6)

translate: /usr/lib/ivl/ivlpp  -v -L -F"/tmp/ivrlg24dc6567c" -f"/tmp/ivrlg4dc6567c" -p"/tmp/ivrli4dc6567c" | /usr/lib/ivl/ivl -v -C"/tmp/ivrlh4dc6567c" -C"/usr/lib/ivl/vvp.conf" -- -
Icarus Verilog Preprocessor version 11.0 (devel) (s20150603-774-g7f95abc6)

Using language generation: IEEE1800-2012,no-specify,xtypes,icarus-misc
PARSING INPUT
LOCATING TOP-LEVEL MODULES
   main
 ... done, 0 seconds.
ELABORATING DESIGN
:0: error: Member el2 of packed union is 4 bits, expecting 0 bits.
1 error(s) during elaboration.
```

I am not too advanced in Systemverilog but from reading the standard and looking at other examples it seems to me the above code should be correct.

One curious thing I noticed it when you take all the struct, enum and union declarations out of the `pkg` package, the error no longer occurs.

I wasn't able to trim down the code any further, for example directly using the enum and logic vector in the `foo_t` union no longer produces the error:

```systemverilog
typedef union packed {
    enum_t el1;
    logic [3:0] el2;
} foo_t;
```

If you need any more information let me know.

I am using the latest commit of iverilog (7f95abc6)

martinwhitaker, 2019-12-03:
This is happening due to the way the compiler handles enumeration type imports. When it sees an imported enumeration type being used, it copies the enumeration type definition into the local scope. However, if the enumeration type is used inside another imported type, the compiler doesn't see it.

I don't know why Steve chose to handle enumeration types in this way. I would have thought they could be handled the same way as any other user-defined type.

You can work round this bug by declaring a dummy variable of type enum_t in your main module.

martinwhitaker, 2019-12-04:
Another example of where the current scheme for handling enum type imports breaks:
````
package p;

  localparam a = 1, b = 2;

  typedef enum { A = a, B = b } enum_t;

endpackage

module a();

  localparam a = 3, b = 4;

  import p::*;

  enum_t e;

  initial begin
    e = A;
    $display(e);
  end

endmodule
````
This should display the value 1, but instead displays the value 3.

patrick96, 2019-12-06:
That's very interesting behavior. Your workaround does the trick.

I still get crashes elsewhere but unfortunately I don't have the time to track those down, I'll have to stick with questa for the time being :/

martinwhitaker, 2019-12-06:
The problem is more widespread than I first thought. This also gives the wrong result:
````
package p;

  localparam a = 1, b = 2;

  typedef logic [b:a] vector_t;

endpackage

module a();

  localparam a = 1, b = 4;

  import p::*;

  vector_t v;

  initial begin:blk
    v = ~0;
    $display("%b", v);
  end
    
endmodule
````
The problem is that a typedef is being elaborated in the context of the scope where it is used, not where it is declared.

martinwhitaker, 2019-12-23:
I've pushed a fix for this to the master branch

closed 2019-12-23 by martinwhitaker

patrick96, 2019-12-23:
Can confirm, I no longer get that error on the newest commit. Thanks! :smiley: