steveicarus/iverilog #290: Function result corruption

zachjs, 2019-12-22:
I have encountered an interesting edge case where two "simultaneous" function calls observe the same result despite being called with different inputs.

The following example:
```SystemVerilog
module top;
    reg [1:0] inp;
    wire [3:0] out;
    Foo foo(inp, out);
    initial begin
        $monitor("inp=%b out=%b", inp, out);
        inp = 0; #10;
        inp = 1; #10;
        inp = 2; #10;
        inp = 3; #10;
    end
endmodule

module Foo(inp, out);
    input wire [1:0] inp;
    output reg [3:0] out;
    always @(inp)
        out[0+:2] = pack(inp + 1);
    always @(inp)
        out[2+:2] = pack(inp + 2);
    function [1:0] pack;
        input reg [1:0] last;
        pack = last;
    endfunction
endmodule
```
Correctly produces the following output:
```
inp=00 out=1001
inp=01 out=1110
inp=10 out=0011
inp=11 out=0100
```

However, if the function is changed to:
```systemverilog
function [1:0] pack;
    input reg [1:0] last;
    begin : block_name
        pack = last;
    end
endfunction
```
The output produced is:
```
inp=00 out=1010
inp=01 out=1111
inp=10 out=0000
inp=11 out=0101
```

I am compiling and running the simulation as follows:
`iverilog test.v && ./a.out`

martinwhitaker, 2019-12-22:
Either behaviour is correct. A non-automatic function shares the same storage over all concurrent function calls, and function calls are not atomic, so the scheduler is allowed to switch between processes between the assignment to the function input  and the assignment to the function output. You can fix your code by declaring `pack` as an automatic function.

N.B. The master branch generates more efficient code for this example, and doesn't show the problem.

zachjs, 2019-12-22:
Thank you for your prompt and thorough clarification!

closed 2019-12-22 by zachjs