steveicarus/iverilog #291: Parameterized class not supported?

apseng03, 2019-12-26:
Hi,
I am having trouble running a simple parameterized class code in Icarus. It does not seem to be supported. Please check. Please see below for the code and commands. 

Command : iverilog -g2012 uvm_param_sclass.sv 

Thanks 
Ani

Error : uvm_param_sclass.sv:4: syntax error
           I give up.

Code:
class uvm_object;
endclass

class uvm_pool #(type KEY=int, T=int) extends uvm_object;
endclass

martinwhitaker, 2019-12-29:
Correct, parameterized classes are not currently supported.

anirudh12032008, 2026-11-13:
I'd like to take this on. Plan is type and value parameters with one specialization per parameter set, building on the class work from #1724/#1770. Will split it into a few PRs.