steveicarus/iverilog #292: Issue in calling class methods

apseng03, 2019-12-27:
Hi,
I am encountering a problem when using class methods. The method call throws an error. Please see below for the error and code.
Thanks,
Ani
  
*Error:* 
iverilog -g2012 class_fn_call.sv

class_fn_call.sv:17: error: No function named `u0.ivl_randomize' found in this context (uvm_driver.ivl_randomize).

*Code:*

class uvm_object;

  int a1;
  virtual function int ivl_randomize();
    a1 = $random();
    $display("%m a1: %0d", a1);
  endfunction : ivl_randomize
endclass : uvm_object

class uvm_driver;
  uvm_object u0;
  virtual function int ivl_randomize();
    int res;
    $display("%m");
    u0 = new();
    // u0.a1 = 33;
    res = u0.ivl_randomize();
  endfunction : ivl_randomize
endclass

module m;
  uvm_driver d0;
  uvm_object u00;
  initial begin
    int res;
    d0 = new();
    res = d0.ivl_randomize();
    u00 = new();
    res = u00.ivl_randomize();
  end
endmodule : m

martinwhitaker, 2020-01-04:
I started to look at this, but kept finding more and more things that are broken. IMHO a rework is required to unify the handling of class properties and normal variables in expressions, then things like this would just work.

svenka3, 2023-01-16:
Thanks for many fixes around class property handling in recent weeks. I believe this issue is still open, I just tried a similar code (not the same as above) and am getting compile error around a class method call. Any thoughts on if this is now doable, please?

Thanks

larsclausen, 2023-01-16:
This is not yet working. We'll get there, but it will take some time. Especially support for virtual methods since that requires changes in vvp as well.