steveicarus/iverilog #295: vcd/fst file rotation

jahagirdar, 2020-01-06:
I am writing a testcase which is expected to run for weeks. At test failure I will be interested only in the previous N cycles of state transitions.
Is there any way to ensure that only the last 'N' cycles of waveform data is retained?

martinwhitaker, 2020-01-06:
There isn't. But with a small change to the VCD writer, you should be able to get close to what you want. This [patch](https://github.com/steveicarus/iverilog/files/4028173/vcd-reopen.patch.txt) modifies the VCD writer to allow you to call $dumpfile after simulation has started, closing the old file and opening the new one. So, in your testbench, do something like this:
````
initial begin  
  $dumpfile("header.vcd");
  $dumpvars;
  forever begin
    $dumpfile("ping.vcd");
    #1000;
    $dumpfile("pong.vcd");
    #1000;
  end
end
````
Then at the end of simulation, if ping.vcd is newer than pong.vcd
````
cat header.vcd pong.vcd ping.vcd > dump.vcd
````
else
````
cat header.vcd ping.vcd pong.vcd > dump.vcd
````
and dump.vcd should contain the last 1000 to 2000 time units of waveform data.

Test this on a short run first!

martinwhitaker, 2020-01-07:
You may want to add a $dumpall after each change of the dump file name, to record the initial state at the start of each fragment.

jahagirdar, 2020-01-16:
Thanks, Is there a similar patch for fst file?

martinwhitaker, 2020-01-16:
No. You could try applying the same principle, but as FST is a compressed format, I'd be surprised if you could simply cut out chunks of the data. You would probably need to do a complete clean start each time you changed the file name.

martinwhitaker, 2020-01-30:
Closing as answered.

closed 2020-01-30 by martinwhitaker