steveicarus/iverilog #296: Shared object name mismatch (for libbz2)

ZenoArrow, 2020-01-12:
I don't know if this is a bug in Icarus Verilog or related to the way the binary I was using was built, but I thought I'd raise it here. On Fedora 31, bzip2-libs has libbz2.so.1 (installed under /usr/lib64 on my laptop), whereas vvp was looking for libbz2.so.1.0  (.0 added to filename). Was able to fix using... sudo cp /usr/lib64/libbz2.so.1 /usr/lib64/libbz2.so.1.0 . Is this an issue with the app being build using a different Linux distro?

martinwhitaker, 2020-01-12:
vvp is linked with the option `-lbz2`. The library version is added by the linker. So yes, it's down to the system where vvp was built.

martinwhitaker, 2020-01-30:
No response, so assuming the answer was satisfactory and closing.

closed 2020-01-30 by martinwhitaker

blackgnezdo, 2024-03-15:
FWIW, a scenario where this also happens is the binaries built on Ubuntu-20 don't work on Rocky-9. The former has this extra `libbz2.so.1.0` which the latter lacks:
```
lrwxrwxrwx 1 root root    15 Sep 18  2022 /lib/x86_64-linux-gnu/libbz2.so.1 -> libbz2.so.1.0.4
lrwxrwxrwx 1 root root    15 Sep 18  2022 /lib/x86_64-linux-gnu/libbz2.so.1.0 -> libbz2.so.1.0.4
-rw-r--r-- 1 root root 74688 Sep 18  2022 /lib/x86_64-linux-gnu/libbz2.so.1.0.4
```
It's a weird state of pointless inconsistency which is nobody's problem to fix yet it is a thorn in users' side.