steveicarus/iverilog #299: procedural continuous assignments are not yet fully supported

daffy1108ch, 2020-01-13:
Hi,

I ran into this problem using 11.0 devel with this code

iverilog -V
Icarus Verilog version 11.0 (devel) (s20150603-558-g11c8262)

Copyright 1998-2017 Stephen Williams

273 function [31:0] lfsr_count;
274   input [31:0] seed;
275   input [31:0] count;
276   reg [31:0] counter;
277   reg fb_xnor ;
278   reg [31:0] lfsr_count;
279   reg [31:0] data_count;
280   begin
281         lfsr_count = seed;
282         for(counter=1; counter<= count; counter=counter+1)
283                 begin
284                         fb_xnor = lfsr_count[30] ^~ lfsr_count[27];
285                         $display("%t : lfsr_count after fb_xnor : fb_xnor = %h, counter = %h, lfsr_count = %h", $time, fb_xnor, counter, lfsr_count);
286                         lfsr_count = {lfsr_count[31-2:0],fb_xnor};
287                         $display("%t : lfsr_count after lfsr_count : fb_xnor = %h, counter = %h, lfsr_count = %h", $time, fb_xnor, counter, lfsr_count);
288 //                      data_count = {lfsr_count, fb_xnor};
289 //                      $display("%t : lfsr_count after data_count : fb_xnor = %h, counter = %h, lfsr_count = %h, data_count = %h", $time, fb_xnor, counter, lfsr_count, data_count);
290                 end
291         end
292 //      assign data_count = {lfsr_count , fb_xnor};
293 //./src/chr_tasks.v:289: tgt-vvp sorry: procedural continuous assignments are not yet fully supported. The RHS of this assignment will only be evaluated once, at the time the assignment     statement is executed.
294 
295         assign data_count = {lfsr_count, fb_xnor};
296         $display("%t : data_count = %h", $time, data_count);
297 endfunction
298 

**./src/chr_tasks.v:295: tgt-vvp sorry: procedural continuous assignments are not yet fully supported.** The RHS of this assignment will only be evaluated once, at the time the assignment statement is executed.

        129060.00 ns : lfsr_count after fb_xnor : fb_xnor = 1, counter = 0000015e, lfsr_count = 7ed23cba
        129060.00 ns : lfsr_count after lfsr_count : fb_xnor = 1, counter = 0000015e, lfsr_count = 7da47975
        129060.00 ns : data_count = fb48f2eb
        129060.00 ns : lfsr_test = 7da47975
TC5 (        129060.00 ns) >     Reading data from register LFSR_DATA (0xc) to check what data word was used for data generation.
TC5 (        142110.00 ns) >       Read 0xfda47975

because the assignment after the for loop should result in

lfsr_test = fda47975 instead of 7da47975

Any workarounds or version which fixes this problem ?

Thanks,

David

daffy1108ch, 2020-01-13:
Here's some additional code showing usage of lfsr_count function

// insert testing logic for lfsr_count
  $display("%t : calculating lfsr using known sent count = %h", $time, reg_data);
  lfsr_test = lfsr_count(32'h12345678, reg_data);
  $display("%t : lfsr_test = %h", $time, lfsr_test);

martinwhitaker, 2020-01-13:
Firstly, since 2009, procedural continuous assignments have been deprecated in the IEEE standard. 1800-2012 has this to say:

> The procedural assign and deassign statements can be a source of design errors and can be an impediment to tool implementation. The procedural assign and deassign statements do not provide a capability that cannot be done by another method that avoids these problems. Therefore, the procedural assign and deassign statements are on a deprecation list.

and indeed I can't see why you are using a procedural continuous assignment here.

Secondly, I don't see what's wrong with the results you report. The final iteration of the for loop assigns `7da47975` to `lfsr_count`. `lfsr_count` is is the return value of the function, so that is the value assigned to `lfsr_test`. And `{31'h7da47975, 1'b1}` evaluates to `32'hfb48f2eb`, so `data_count` contains the correct value too.

martinwhitaker, 2020-01-30:
No response, so closing as invalid. Feel free to reopen if you disagree and can explain why the current behaviour is wrong.

closed 2020-01-30 by martinwhitaker