steveicarus/iverilog #301: unexpected behaviour change since v10.2

akukulanski, 2020-01-25:
Hi,
Since v10.2 one further simulation step seems to be required for continuous assingments inside *always* procedure, and not for the same logic described with *assign*.
Link to testcase: https://github.com/akukulanski/iverilog-testcase

Testcase description:
In the repo there are two tests which are the same, except that one has an initial extra clock delay before reading the output. In iverilog 10.1 both tests pass, in v10.2 only the one with the extra delay. The failure is because the output signal has undetermined value in the first read.

Now, if I change this:
```verilog
reg x;
always @* begin
    x = 1'h0;
    casez (y )
      1'h0:
          x = 1'h1;
      1'h1:
          x = 1'h0;
    endcase
end
```

To this: (which describes the same logic)
```verilog
wire x;
assign x = (y == 1'h0) ?
           1'h1 : 1'h0;
```
It works and both test PASS with **both versions**.

Is this a bug or am I missing something?

Thanks!

martinwhitaker, 2020-01-25:
It seems likely there is a race in your logic, but without a Verilog testbench that demonstrates the problem, I can't tell. At what simulation time does the first clock edge occur?

akukulanski, 2020-01-25:
The sim time there 0.0.
I'd think the same, but I find weird that it started to happen since v10.2 and that it depends on how I write the HDL remaining the logic described the same. That makes me think that there is some difference on how the verilog is parsed, but I really don't know. Any idea?

martinwhitaker, 2020-01-26:
Does this still occur with v10.3 and with the latest code from git?

akukulanski, 2020-01-27:
Yes, it's the same with v10.3

martinwhitaker, 2020-01-27:
The explanation is that cocotb compiles the code with the -g2012 option, enabling SystemVerilog. The SystemVerilog standard specifies that variable initialisations, e.g. `reg y = 1'h0` , are performed before the start of simulation, i.e. before any `initial` or `always` process is executed. Before v10.2, Icarus did not do this, instead conforming to the traditional Verilog behaviour of initialising variables at time 0.

Why does this matter? Well, consider this version of your code, with a clock generator added
````
`timescale 1ns/1ps

module top;

reg clk = 0;

always begin
    clk = ~clk;
    #1;
end

reg x;
reg y = 1'h0;

always @(posedge clk)
    y  <= y + 1'h1;

always @* begin
    x = 1'h0;
    case (y)
      1'h0:
          x = 1'h1;
      1'h1:
          x = 1'h0;
    endcase
end

initial begin
  $monitor($time,,clk,,y,,x);
  #10 $finish;
end

endmodule
````
When compiled and run with the -g2012 option, this will show the value of `x` remains at `1'bx` until time 2. This is because the clock generator process gets started before the process that toggles `y`, so `y` doesn't get toggled at time 0. And because `y` is initialised before simulation starts, there is nothing else to trigger the process that updates `x`.

If compiled and run without the -g2012 option, `y` is initialised at time 0, which triggers the process that updates `x`, setting `x` to `1'b1`.

Continuous assignments are always evaluated at time 0, hence when you replace the process that updates `x`with a continuous assignment, the behaviour is the same as compiling without the -g2012 option.

akukulanski, 2020-01-28:
That was very clear, I'm closing this issue.
Thanks a lot Martin! :)

closed 2020-01-28 by akukulanski