steveicarus/iverilog #306: Segmentation fault when ``iverilog -g1995 -S -o counter counter.v``

vowstar, 2020-02-04:
## Brief

Coredump may occur with ``iverilog -g1995 -S -o counter counter.v``

```bash
iverilog -g1995 -S -o counter counter.v
sh: line 1: 154562 Done                    /usr/lib64/ivl/ivlpp -L -F"/tmp/ivrlg25367e8a2" -f"/tmp/ivrlg5367e8a2" -p"/tmp/ivrli5367e8a2"
     154563 Segmentation fault      (core dumped) | /usr/lib64/ivl/ivl -C"/tmp/ivrlh5367e8a2" -C"/usr/lib64/ivl/vvp-s.conf" -- -
```

## Code

File name: ``counter.v``

```verilog
module counter(out, clk, reset);

  parameter WIDTH = 8;

  output [WIDTH-1 : 0] out;
  input 	       clk, reset;

  reg [WIDTH-1 : 0]   out;
  wire 	       clk, reset;

  always @(posedge clk)
    out <= out + 1;

  always @reset
    if (reset)
      assign out = 0;
    else
      deassign out;

endmodule // counter

```

## Log

run ``iverilog -g1995 -S -o counter counter.v``

```
sh: line 1: 149801 Done                    /usr/lib64/ivl/ivlpp -L -F"/tmp/ivrlg27362106f" -f"/tmp/ivrlg7362106f" -p"/tmp/ivrli7362106f"
    149802 Segmentation fault      (core dumped) | /usr/lib64/ivl/ivl -C"/tmp/ivrlh7362106f" -C"/usr/lib64/ivl/vvp-s.conf" -- -

```

ltrace
```
memcpy(0x55d3b76b174a, " -C"/usr/lib64/ivl/vvp-s.conf"\0", 31)                                                                                                            = 0x55d3b76b174a
realloc(0x55d3b76b16c0, 174)                                                                                                                                              = 0x55d3b76b16c0
memcpy(0x55d3b76b1768, " -- -\0", 6)                                                                                                                                      = 0x55d3b76b1768
system("/usr/lib64/ivl/ivlpp  -L -F"/tmp"...sh: line 1: 150423 Done                    /usr/lib64/ivl/ivlpp -L -F"/tmp/ivrlg228852716" -f"/tmp/ivrlg28852716" -p"/tmp/ivrli28852716"
    150424 Segmentation fault      (core dumped) | /usr/lib64/ivl/ivl -C"/tmp/ivrlh28852716" -C"/usr/lib64/ivl/vvp-s.conf" -- -
<no return ...>
--- SIGCHLD (Child exited) ---
<... system resumed> )                                                                                                                                                    = 35584
getenv("IVERILOG_ICONFIG")                                                                                                                                                = nil
remove("/tmp/ivrlg28852716")      
```

strace
```
rt_sigprocmask(SIG_BLOCK, ~[], [CHLD], 8) = 0
clone(child_stack=0x7f7f0eb6bff0, flags=CLONE_VM|CLONE_VFORK|SIGCHLD) = 148997
munmap(0x7f7f0eb63000, 36864)           = 0
rt_sigprocmask(SIG_SETMASK, [CHLD], NULL, 8) = 0
wait4(148997, sh: line 1: 148998 Done                    /usr/lib64/ivl/ivlpp -L -F"/tmp/ivrlg21079ea71" -f"/tmp/ivrlg1079ea71" -p"/tmp/ivrli1079ea71"
    148999 Segmentation fault      (core dumped) | /usr/lib64/ivl/ivl -C"/tmp/ivrlh1079ea71" -C"/usr/lib64/ivl/vvp-s.conf" -- -
[{WIFEXITED(s) && WEXITSTATUS(s) == 139}], 0, NULL) = 148997
rt_sigaction(SIGINT, {sa_handler=SIG_DFL, sa_mask=[], sa_flags=SA_RESTORER, sa_restorer=0x7f7f0e977480}, NULL, 8) = 0
rt_sigaction(SIGQUIT, {sa_handler=SIG_DFL, sa_mask=[], sa_flags=SA_RESTORER, sa_restorer=0x7f7f0e977480}, NULL, 8) = 0
rt_sigprocmask(SIG_SETMASK, [], NULL, 8) = 0
--- SIGCHLD {si_signo=SIGCHLD, si_code=CLD_EXITED, si_pid=148997, si_uid=1000, si_status=139, si_utime=0, si_stime=0} ---
unlink("/tmp/ivrlg1079ea71")            = 0
unlink("/tmp/ivrlh1079ea71")            = 0
unlink("/tmp/ivrlg21079ea71")           = 0
unlink("/tmp/ivrli1079ea71")            = 0
exit_group(139)                         = ?
+++ exited with 139 +++
```

steveicarus, 2020-02-04:
Looks like it is the -S flag that triggers this. It's probably an old bug, because I see it with v10 and the devel branches.

martinwhitaker, 2020-02-04:
I last saw that style of modelling an asynchronous reset flip-flop more than 20 years ago!

The compiler should not crash of course, but the fix will most likely be to report it as unsynthesisable. If you really need synthesis, you can recode it in the more usual style:
````
always @(posedge clk or posedge reset) begin
  if (reset)
    out <= 0;
  else
    out <= out + 1;
end
````

steveicarus, 2020-02-04:
Procedural continuous assignments are not the way to do things these days, that is true, and systhesis should certainly fail, but the way synthesis fails in icarus Verilog is to generate procedural code instead. Certainly, the crash is wrong.

But Martin is correct, the example code is considered poor form these days, and might not even work with many synthesizers.

vowstar, 2020-02-05:
> I last saw that style of modelling an asynchronous reset flip-flop more than 20 years ago!
>
> The compiler should not crash of course, but the fix will most likely be to report it as unsynthesisable. [...]

You are right, the test code from https://iverilog.fandom.com/wiki/Getting_Started

steveicarus, 2020-02-05:
Well I will certainly fix the example on the documentation page;-)

martinwhitaker, 2020-02-09:
Fixed in both the master and v10 branches. I didn't find an easy way to generate procedural code instead, so it now outputs a "sorry" message.

I've also revised the counter example on the wiki to use the synthesisable style.

closed 2020-02-09 by martinwhitaker