steveicarus/iverilog #307: Are struct fields not assignable?

cpitclaudel, 2020-02-05:
As a follow-up to #161, the following example fails.  I'm using a copy of the master branch built today (`Icarus Verilog version 11.0 (devel) (b111476)`)

```systemverilog
typedef struct packed {
   logic b;
} single_bit;

typedef struct packed {
   single_bit b1;
   single_bit b2;
} two_bits;

module simple(input two_bits b2in,
              output two_bits b2out);
   assign b2out.b1.b = b2in.b1.b;
endmodule
```

Here's the output:

```
$ /build/iverilog/install/bin/iverilog -delaborate -g2012 -gxtypes simple.v
debug: Enable elaborate debug
<toplevel>: elaborate: elaboration work list done. Start processing residual defparams.
<toplevel>: elaborate: Start calling Package elaborate_sig methods.
-:1: PPackage::elaborate_sig: Start package scope=$unit
-:1: PPackage::elaborate_sig: Done package scope=$unit, flag=1
<toplevel>: elaborate: Start calling $root elaborate_sig methods.
<toplevel>: elaborate: Start calling root module elaborate_sig methods.
<toplevel>: debug: simple: port elaboration root 2 ports
simple.v:10: PWire::elaborate_sig: Signal b2in, wtype=wire, data_type_=logic, is_implicit_scalar=true
simple.v:10: debug: Create signal wire 2 bit packed struct b2in in scope simple
simple.v:11: PWire::elaborate_sig: Signal b2out, wtype=reg, data_type_=logic, is_implicit_scalar=true
simple.v:11: debug: Create signal reg 2 bit packed struct b2out in scope simple
simple.v:10: PEIdent::elaborate_subport: path_ = "b2in", unpacked_dimensions=0, port_type()=PINPUT
<toplevel>: debug: simple: adding module port b2in
simple.v:11: PEIdent::elaborate_subport: path_ = "b2out", unpacked_dimensions=0, port_type()=POUTPUT
<toplevel>: debug: simple: adding module port b2out
simple.v:12: PEIdent::elaborate_lnet_common_: Symbol not found, try again with path_prefix=b2out.b1 and method_name=b
simple.v:12: error: Net b2out.b1.b is not defined in this context.
<toplevel>: debug:  finishing with 1 root scopes 
1 error(s) during elaboration.
```

Thanks a lot for your work on iverilator.

martinwhitaker, 2020-02-08:
The fixes done for procedural assignments to nested struct fields in elab_lval.cc need to be replicated for continuous assignments in elab_net.cc.

steveicarus, 2020-02-09:
Since I did the elab_lval.cc fixes, I can look into doing this fix as
well. May take some time because day job takes a lot of time these
days.

cpitclaudel, 2020-02-09:
> Since I did the elab_lval.cc fixes, I can look into doing this fix as well.

Thanks a lot!

steveicarus, 2020-11-30:
See PR#401.

closed 2020-11-30 by steveicarus

cpitclaudel, 2020-11-30:
Wonderful, thanks a lot!

fogjava, 2020-12-01:
thanks a lot iverilator