steveicarus/iverilog #310: Trailing comma in argument list leads to confusing error

meawoppl, 2020-02-12:
```
module Foo(
    input bar,
);
endmodule
```
Will give the error
`Foo.v:3: error: NULL port declarations are not allowed.`

Not a real problem, but it took me a while to figure out. I think the trailing comma _might_ be valid verilog, but I am not a language expert.

Millennium-Software-Design, 2020-02-12:
On 2/12/20 11:59 AM, Matty G wrote:
> |module Foo( input bar, ); endmodule |
> 
> Will give the error
> |Foo.v:3: error: NULL port declarations are not allowed.|
> 
> Not a real problem, but it took me a while to figure out. I think the 
> trailing comma /might/ be valid verilog, but I am not a language expert.

I've never seen a verilog simulator happy with a trailing comma.

Bryan
https://msd.llc/

meawoppl, 2020-02-13:
It appears to pass muster on the lint pass, which to me points at a probable glitch in the parser grammer somewhere.  Not an urgent bug IMHO, but one that will foul up n00b's like myself.

martinwhitaker, 2020-02-13:
From IEEE 1364-2005:
````
list_of_port_declarations ::=
( port_declaration { , port_declaration } )
| ( )

port_declaration ::=
{attribute_instance} inout_declaration
| {attribute_instance} input_declaration
| {attribute_instance} output_declaration

input_declaration ::= input [ net_type ] [ signed ] [ range ]
list_of_port_identifiers

list_of_port_identifiers ::= port_identifier { , port_identifier }
````
so your code is illegal syntax, and whatever lint tool you are using should report that.

The error message could perhaps be improved. Writing NULL in capitals is a bit misleading.

meawoppl, 2020-02-13:
Concur. I think the ide plugin I am using has a bug. Ditto yosys the
synthesis tool.

martinwhitaker, 2020-02-15:
I have improved the error message in the master branch.

closed 2020-02-15 by martinwhitaker