steveicarus/iverilog #315: specify section params ignored where conditionals are present

Johnlon, 2020-03-31:
I am finding that under certain conditions the specify timings are not respected.

This appears to the be the case if a continuous assignment contains a conditional.
In my case ternary operators cause the timings to be ignored.

I use the current source code build using options .. "-Wall -g2012 -gspecify -Ttyp"

However, this behaves the same  - see runnable example ...  https://www.edaplayground.com/x/hDa

Cross posted here : https://stackoverflow.com/questions/60938727/icarus-verilog-specify-delays-not-respected-if-there-are-conditionals

A response there is that this issue doesn't happen on two other simulators that were tried.

So the suggestion is that it's a bug in icarus.

=====

`// Code your design here
`timescale 1ns/1ns

module hct74245(
  input dir,
  input nOE,
  inout [7:0] A,
  inout [7:0] B,
  output [7:0] C // Added to demonstrate that a straight dependency timing works correctly
);

  // HCT typical @ 5v according to https://assets.nexperia.com/documents/data-sheet/74HC_HCT245.pdf

  specify
    (A *> C) = 100; // This delay works OK
    (A *> B) = 10;  // The rest of these delays do not work
    (B *> A) = 10; // not respected
    (dir *> A) = 16;// not respected
    (dir *> B) = 16;// not respected
    (nOE *> A) = 16;// not respected
    (nOE *> B) = 16;// not respected
  endspecify

  assign A=nOE? 8'bzzzzzzzz : dir?8'bzzzzzzzz:B;
  assign B=nOE? 8'bzzzzzzzz : dir?A:8'bzzzzzzzz;
  assign C=A; // THIS IS DELAYED BY 100

  // HAVE TO USE THIS APPROACH TO MAKE TIMINGS WORK AT ALL
  // assign #16 A=nOE? 8'bzzzzzzzz :dir?8'bzzzzzzzz:B;
  // assign #16 B=nOE? 8'bzzzzzzzz :dir?A: 8'bzzzzzzzz;

endmodule: hct74245
`

Test harness which prints ERROR where the timings are not respected.

`/* BUG? Known limitation??

Apart from the A->C timing which is direct (no conditionals) the specify timings are no respected.
I have to add a inline delay to get anything.

It seems that specify delays are ignored where there's a conditional. 
*/

`timescale 1ns/1ns

module tb();

  tri [7:0]A;
  tri [7:0]B;

  tri [7:0]C; // 'C' IS NOT PART OF ORIGINAL DESIGN - HOWEVER THIS TIMING IS RESPECTED COS THERE ARE NO CONDITIONALS IN THE ASSIGNMENT

  logic [7:0] Vb=8'b00000000;
  logic [7:0] Va=8'b11111111;

  logic dir;
  logic nOE;

  assign B=Vb;
  assign A=Va;

  hct74245 buf245(.A, .B, .dir, .nOE);

  integer timer;

  initial begin
    $display("disable output , set dir a->b");
    dir <= 1; // A->B
    nOE <= 1;
    Va=8'b11111111;
    Vb=8'bzzzzzzzz;

    #50 // time to settle

    // NOW throw outputs on and time how long it takes for expected output to appear.
    
    // It should take 16 to propagate from nOE -> B but the change is instantaneous

    $display("enable output - B will change immediately");

    timer=$time;
    nOE <= 0;
    wait(B === 8'b11111111);
    if ($time - timer < 16)
      $display("%6d", $time, " ERROR TOO QUICK - EXPECTED nOE->B = 16ns - BUT TOOK %-d", ($time - timer));
    else
      $display("%6d", $time, " OK %-d", ($time - timer));

    #50 // settle

    // Change data in - This should take 10 to propagate A->B but is instantaneous
    $display("change A - B will change immediately (but C is delayed as expected by 100)");

    Va=8'b00000000;

    timer=$time;
    nOE <= 0;
    wait(B === 8'b00000000);
    if ($time - timer < 10)
      $display("%6d", $time, " ERROR TOO QUICK - EXPECTED A->B = 10ns - BUT TOOK %-d", ($time - timer));
    else
      $display("%6d", $time, " OK - TOOK %-d", ($time - timer));

  end
endmodule
`

martinwhitaker, 2020-04-02:
The problem was that the vvp target code generator only output path timing information for nets with a single driver. I've now added support  for multiply-driven nets. Fixed in both the master and v10 branches.

closed 2020-04-02 by martinwhitaker

Johnlon, 2020-04-02:
confirmed fixed - on master

thanks