steveicarus/iverilog #316: latest master seems to have affected pulldowns

Johnlon, 2020-04-11:
See demo of  a test that works on old verilog: https://www.edaplayground.com/x/kD7

I reported an earlier problem relating to the delays in the circuit not being respected.
That was fixed in master. https://github.com/steveicarus/iverilog/issues/193

----

However, for the same circuit I've now noticed a change in behaviour.

The test below fails.
It used to pass.

It looks like the pull down to 0 causes a conflict with the H outputs of my buffer which is not what I expect a pull down to do - am I misusing it?

FTI I think there is stil something weird going on with those timings

`
`timescale 1ns/100ps
`default_nettype none

module hct74245( 
    input dir,
    input nOE,
    inout [7:0] A,
    inout [7:0] B
);
    // HCT typical @ 5v according to https://assets.nexperia.com/documents/data-sheet/74HC_HCT245.pdf
    /*
        specify
            (A => B) = (10);
            (B => A) = (10);
            (dir *> A) = (16);
            (dir *> B) = (16);
            (nOE *> A) = (16);
            (nOE *> B) = (16);
        endspecify
    */

    assign A=nOE? 8'bzzzzzzzz :dir?8'bzzzzzzzz:B;
    assign B=nOE? 8'bzzzzzzzz :dir?A:8'bzzzzzzzz;

    always @(*)
        $display("%6d", $time,  " BUF : A=%8b ", A, "B=%-8b ", B, "dir=%1b", dir, " nOE=%1b", nOE);
   
endmodule: hct74245

// THE FIRST TEST FAILS ON MASTER

module test();

    wire AtoB=1'b1;     // direction A to B

    logic nOEvar=0;      // output enabled
    wire nOE=nOEvar;      // output enabled

    // I expect this same value on B !!!!!!
    wire [7:0] A = 8'b10101010;       

    // NOTE : Remove the pulldown and change this from tri to tri0 and then the test passes
    tri  [7:0] B;                                   
    
     // transceiver - B should either follow A or be pulled low 
     hct74245 bufTest(.A, .B, .dir(AtoB), .nOE); 

    // pull B to 0 when output of buffer is disabled.
    // NOTE:  Remove this pulldown and change the B wire from tri to tri0 and then the test passes
    pulldown pullTo0[7:0](B);

    always @(*) 
        $display("%6d", $time,  " TEST: A=%8b ", A, "B=%-8b ", B);

    initial begin

        // output B should follow input A because nOE is low
        nOEvar=1'b0;      // output enabled
        #100
        if (B !== 8'b10101010) begin 
            $display("FAILED 1: '%b' is not '%b' ", B, 8'b10101010);
            $finish_and_return(1); 
        end

        // output B should be 0 because nOE is high, and B should be Z therefore pulldown should pull to 0
        nOEvar=1'b1;      // output disabled
        #100
        if (B !== 8'b00000000) begin 
            $display("FAILED 2: '%8b' is not '%8b' ", B, 8'b00000000);
            $finish_and_return(1); 
        end
    end
endmodule : test
`

Johnlon, 2020-04-11:
Is there an automated test suite that this example could be added to?

Johnlon, 2020-04-11:
Also is there meant to be a functional difference between using "tri0" vs using a "pulldown"?

martinwhitaker, 2020-04-12:
There should be no functional difference between using tri0 vs. using a wire/tri with an attached pulldown0. However the compiler front-end preserves the difference, which sends the target code generator down a different path.

The compiler test suite can be found at https://github.com/steveicarus/ivtest. New regression tests are added whenever a bug is fixed.

Johnlon, 2020-04-13:
Some opinion on subtle diffs...

https://stackoverflow.com/questions/61162933/systemverilog-tri0-vs-pulldown-is-there-any-difference/61164942#61164942

martinwhitaker, 2020-05-07:
This is now fixed. Please check it has not introduced any more regressions in your code.

caryr, 2020-05-08:
This is failing under Ubuntu 16.04, but passes for 18.04.

For 16.04 gcc and clang compiles both fail. Send me an email if you want me to debug further.

martinwhitaker, 2020-05-08:
It builds and runs fine here in a fresh install of Ubuntu 16.04 in a VM (using gcc).

caryr, 2020-05-09:
Okay, thanks for checking. I'll try to figure out what is broken in my environment. Probably something I'm working on interfering some how.

caryr, 2020-05-10:
Figured the issue out on my end. Sorry for the noise.

Johnlon, 2020-05-11:
will try

Johnlon, 2020-05-11:
Dont know if it significant but during make I got non fatal message

g++ -DHAVE_CONFIG_H -I. -Ilibmisc  -Wall -Wextra -Wshadow   -g -O2 -MD -c parse.cc -o parse.o
parse.cc:8458:18: warning: missing initializer for member ‘vlltype::text’ [-Wmissing-field-initializers]
   = { 1, 1, 1, 1 }
                  ^

Johnlon, 2020-05-11:
Is there a label for this version?

martinwhitaker, 2020-05-11:
The warning comes from code generated by flex/bison, so we can't fix that.
We only create tags for formal releases and snapshots. master is pretty stable these days, so there hasn't been a snapshot for a long time. If you want an informal label, use `git describe --tags`.

martinwhitaker, 2020-06-01:
No further feedback, so closing as fixed.

closed 2020-06-01 by martinwhitaker