steveicarus/iverilog #318: backslash-newline in Xilinx Vivado generated code cause iverilog to fail (inline attributes)

lisper, 2020-04-12:
Trying to simulate verilog source generated by Xilinx Vivado. The code has in-line attributes which have a line break/continuation in the middle - otherwise known as a "backslash-newline".

This seems to confuse iverilog.  here's an example.  Note how the attribute "NUM_WRITE_THREADS" is broken in half by the backslash-newline.

vivado seems to want to keep these lines to a certain length, sadly.

...
output wire mem_axi_rready;
(* X_INTERFACE_PARAMETER = "XIL_INTERFACENAME mem_axi, DATA_WIDTH 32, PROTOCOL AXI4LITE, FREQ_HZ 100000000, ID_WIDTH 0, ADDR_WIDTH 32, AWUSER_WIDTH 0, ARUSER_WIDTH 0, WUSER_WIDTH 0, RUSER_WIDTH 0, BUSER_WIDTH 0, READ_WRITE_MODE READ_WRITE, HAS_BURST 0, HAS_LOCK 0, HAS_PROT 1, HAS_CACHE 0, HAS_QOS 0, HAS_REGION 0, HAS_WSTRB 1, HAS_BRESP 1, HAS_RRESP 1, SUPPORTS_NARROW_BURST 0, NUM_READ_OUTSTANDING 1, NUM_WRITE_OUTSTANDING 1, MAX_BURST_LENGTH 1, PHASE 0.000, CLK_DOMAIN top_level_block_partial_clk, NUM_READ_THREADS 1, NUM_WRI\
TE_THREADS 1, RUSER_BITS_PER_BYTE 0, WUSER_BITS_PER_BYTE 0, INSERT_VIP 0" *)
(* X_INTERFACE_INFO = "xilinx.com:interface:aximm:1.0 mem_axi RDATA" *)
...

caryr, 2020-05-03:
I have looked at the standard (1800-2012) and I cannot find where this is supported for a string literal. This looks to be invalid syntax. I believe line continuation is only supported for macros. Can you create a simplified input line that demonstrates the issue so we can review it. I will look at other versions of the standard to see if this was supported in the past. My guess is they are using syntax not covered by the standard. It is probably not too hard to add this, but I want to make sure **if this is added** it actually solves your problem.

caryr, 2020-05-09:
Please provide more information as previously requested or this will eventually be closed as invalid.

caryr, 2020-06-01:
No information provided so closing this.

closed 2020-06-01 by caryr