steveicarus/iverilog #321: Iverilog hangs on testbench

meheff, 2020-05-07:
Running the attached Verilog testbench results in iverilog hanging. The curious thing is that hitting control-C to break the simulation, then continuing it sometimes (non-deterministically) results in the simulation continuing correctly to completion. A commercial simulator from Metrics does not hang.

Repro:

$ iverilog hang_test.v.txt -o /tmp/hang_test.v.out && vvp /tmp/hang_test.out
$ iverilog -v
Icarus Verilog version 10.3 (stable) ()

[hang_test.v.txt](https://github.com/steveicarus/iverilog/files/4595582/hang_test.v.txt)

meheff, 2020-05-07:
Correction, I tested with the latest devel version (also hangs with the old version installed on the system too):

$ iverilog  -v
Icarus Verilog version 11.0 (devel) (s20150603-836-g58eb2023)

martinwhitaker, 2020-05-07:
It's most likely a subtle race in your code. Compile with
````
iverilog -pfileline=1 ...
````
and you will see the lines of code it is looping round, which should help you identify the problem.

meheff, 2020-05-07:
Thanks! I'll have a look.

meheff, 2020-05-08:
I dove into this a bit, and have a question. In the Verilog I commonly use a construct like so:

```
  always @ (*) begin
    foo = 0;
    case (state)
      State0: begin
        foo = 1;
      end
      State1: begin
        // foo not assigned so will get value 0.
      end
      ....
    end
  end
```

'foo' is assigned a default value prior to the case statement, and then sometimes overridden in some of the case blocks. This assignment to 0, then 1 in the same evaluation of the always block looks like it triggers the blocks with a sensitivity list that includes 'foo' even if the value of 'foo' entering the always block is the same as it is leaving (foo == 1). That is, the always block does not actually change the value of foo.

This looks like it results in infinite looping during simulation at least with iverilog. (I don't know why Ctl-C followed by continue sometime fixes it, but that may be another issue). Is this expected behavior?

A work around it to only assign a value once by sinking the default assignment (foo = 0) into the cases, though this can be a little error prone if the code includes assignments within deep conditionals as you have to ensure exactly one single assignment along all code paths.

Thanks!

meheff, 2020-05-08:
Here's a minimal repo.  The assignments out_reg=0 followed by out_reg=1 in the two instantiated modules m0 and m1 end up triggering each others' always blocks ad infinitum though there are no combinational loops.

```
module m(
         input  trigger0,
         input  trigger1,
         output out);
   reg          out_reg;
   reg          tmp;
   always @ (*) begin
      // tmp assignment only here to add trigger0 and trigger1 to sensitivity
      // list of always block. tmp is dead.
      tmp = trigger0 | trigger1;
      
      out_reg = 0;
      out_reg = 1;
   end
   wire out;
   assign out = out_reg;
endmodule   

module tb;

   reg trigger;
   wire tmp0;
   wire tmp1;

   m m0 (
         .trigger0(trigger),
         .trigger1(tmp0),
         .out(tmp1)
         );

   m m1 (
         .trigger0(trigger),
         .trigger1(tmp1),
         .out(tmp0)
         );
         
   initial begin
      // Kick things off by assigning to trigger.
      trigger = 1;
   end
endmodule      
```

caryr, 2020-05-09:
This is certainly a race. How is a transition from 0 to 1 not considered an edge? This style of coding is usually used for FSM next state calculations which is clocked so not impacted by glitches. Verilog may run blocks in order, but it dos not treat them as atomic operations.

martinwhitaker, 2020-05-09:
As I read it, this is non-deterministic behaviour permitted by the standard. Consider this simple example:
````
reg value = 0;
integer count = 0;

always @(value) count = count + 1;

initial begin
  #1;
  value = 1;
  value = 0;
end
````
Ignoring the time-0 behaviour, the first assignment to `value` generates an update event. This causes an evaluation event for the `@(value)` to be added to the active event queue. At this point the simulator has two choices: evaluate the `@(value)`, or execute the `value = 0`. If it chooses the first course of action, an edge will be detected. If it chooses the second, when the `@(value)` is finally evaluated, `value` will have returned to 0, and no edge will be detected.

caryr, 2020-05-10:
Actually there's even more non-deterministic behaviour in this example Martin. It could trigger once or twice. Once when it goes 0->1 and once again when it goes from 1->0. I believe Icarus always finishes running as much of the currently running block it can before transitioning to other blocks. So this example will only trigger once. The 0->1 transition creates a trigger that is still active when the value transitions from 1->0. In the case of a parallel simulation or one using a different algorithm this could trigger twice.

I believe there is a VCD bug report related to this. 0->1->0 triggers a VCD update even though the value does not change.

Also I think the non-determinism is if the always triggers once or twice not once or not at all. A simulator could conceptually filter non-blocking assignments, but Icarus does not and I doubt others do either.

For example:
```
module top;
  bit trig;

  always @(posedge trig) $display("Had a positive edge");
  always @(negedge trig) $display("Had a negative edge");
  always @(trig) $display("Trigger value: %d", trig);

  initial begin
    #1;
    trig = 1'b1;
    trig = 1'b0;
    $display("Done with initial block");
  end
endmodule
```
returns
```
Done with initial block
Had a positive edge
Trigger value: 0
Had a negative edge
```

caryr, 2020-05-10:
Sorry make that blocking assignments.

martinwhitaker, 2020-05-10:
@caryr , indeed, I was only trying to explain the non-determinism, not highlight all that could go wrong with that code.

@meheff, but rereading your minimal example, I think the important difference is that you are passing the signals through an intermediate wire and module port. For the same reason I gave above, the wire may be updated before or after the second assignment. Icarus updates the wire before the second assignment, so a glitch gets through. Other simulators update the wire after the second assignment, so the glitch gets filtered out.

Try this even more minimal example:
````
module tb;

  reg trigger;
  reg tmp0;
  reg tmp1;

  always @(trigger or tmp1) begin
    $display("block 0 triggered at time %t", $time);
    tmp0 = 0;
    tmp0 = 1;
  end

  always @(tmp0) begin
    $display("block 1 triggered at time %t", $time);
    tmp1 = 0;
    tmp1 = 1;
  end

  initial begin
    // Kick things off by assigning to trigger.
    #1 trigger = 1;
  end
endmodule
````
Running that on EDA playground using the Aldec simulator shows an infinite loop.
Now change it to
````
module tb;

  reg trigger;
  reg tmp0r;
  reg tmp1r;

  wire tmp0 = tmp0r;
  wire tmp1 = tmp1r;

  always @(trigger or tmp1) begin
    $display("block 0 triggered at time %t", $time);
    tmp0r = 0;
    tmp0r = 1;
  end

  always @(tmp0) begin
    $display("block 1 triggered at time %t", $time);
    tmp1r = 0;
    tmp1r = 1;
  end

  initial begin
    // Kick things off by assigning to trigger.
    #1 trigger = 1;
  end
endmodule
````
Using the Aldec simulator, it is no longer an infinite loop. Using Icarus, it still is.

martinwhitaker, 2020-06-01:
During a discussion about scheduling semantics on iverilog-devel, it was pointed out that `@(...) begin ... end` was in fact a single statement with a timing control, and hence any update events triggered by execution of the `begin ... end` block should not be executed until after the last statement in the block has been executed. In light of this, I now think that the Icarus behaviour is incorrect.

steveicarus, 2020-12-06:
How is this not a race and undefined behavior? Sure, blocks run to completion, but that doesn't change the fact that the blocks i the example still trigger events.

martinwhitaker, 2020-12-06:
Consider the second example in my comment from 10th May. In the first always construct, the first assignment to tmp0r adds an evaluation event for the continuous assignment to the active event queue. But if the block has to run to completion before another event is processed, that evaluation will not occur until after the second assignment to tmp0r, so by the time it is evaluated, tmp0r has returned to 0, so the expression value remains unchanged and doesn;t generate an update event.

caryr, 2020-12-06:
First to clarify I see the VCD behavior as orthogonal since I think it should be related to has the value changed when we get to the postponed region not has the value changed in this time step. Even this is not 100% clear as can be shown in this text from 1800-2012-21.7.2.1

> Only the variables that change value during a time increment are listed.

I choose to interpret this as actually changing value at the postponed region and that is mostly because it is what makes most sense. To fix this we likely need $past() like functionality which effectively doubles the data size for everything being dumped unless we can figure out how to only create $past() functionality for events that change in the current time step. Basically a callback that fires just before a variable/net is updated so we can save the previous value in the dumper. cbValueChange is what we use now and that fires after the change. Maybe we really need a dumper specific value change callback that returns both the current and new value. It would then become the dumpers responsibility to save the first current value of  time step and use that to filter the final output.

Martin,

Can you provide a reference to a place in the standard where it specifies this functionality? What I find in 1800-2012-4.3 is the following:

> Every change in state of a net or variable in the system description being simulated is considered an update event.

Which implies the current behavior is correct. I could not find anything that implied begin/end pairs create some kind of atomic behavior with regard to variable updates. The only place I assign to the same variable in a block is for next state calculations in a FSM and since that is only used at a clock edge these spurious events are of no consequence.

caryr, 2020-12-06:
Also on the VCD functionality I could argue the current behavior is good since normal dumpers should just do the right thing and an enhanced one could use these redundant events to indicate a zero time glitch to the user. Maybe I should ask Tony about adding support for this to GTKWave.

steveicarus, 2020-12-06:
There is this example:
```
module main;

   reg flag;
   
   always @(flag) begin
      $display("At %0t, Flag=%b", $time, flag);
   end

   always #50 begin
      flag = 0;
      flag = 1;
   end
   initial #500 $finish;
   
endmodule // main
```
Note that the assignment to `flag` generates an update event, even if the update is, ultimately, null since the always statement runs atomically. The open question is whether the event should be generated or not, and that would require a dive into transient elimination.

martinwhitaker, 2020-12-06:
Do you consider a statement to be be an evaluation event or an update event? The IEEE standard never makes that clear.

caryr, 2020-12-06:
I believe my quote from the standard above makes it clear this should be considered an update event.

martinwhitaker, 2020-12-06:
@caryr, but consider this snippet from the simulation reference model:
````
  E = any active event;
  if (E is an update event) {
    update the modified object;
  } else { /* shall be an evaluation event */
    evaluate the process;
    add update events to the event queue;
  }
````
An initial or always construct creates a process that executes statements, so from that snippet I would say that execution of a statement is part of an evaluation event (some of which may add update events to the event queue). But I can find other places in the standard which appear to contradict that.

Trying to determine what is correct by reading the standard is dicey at best. I remind you of this comment from when we were discussing this on iverilog-discuss:

https://sourceforge.net/p/iverilog/mailman/message/37013421/

caryr, 2020-12-06:
FYI the code Steve provided produces the following when running on one of the big three simulators:

> At 50, Flag = 1

Now if you add a $display("Zero time glitch at %0t", $time); between the two assignments to "flag" this triggers every 50 timeunits. If the $display() is before or after the assignments there is only a single trigger. This difference is likely an issue in their optimizer. I think we would need to have this checked on other simulators to get a consensus because this seems like an optimization they are doing to be faster, but I'm not certain it is correct.

This is copied from 1800-2012 (also at least visually matches 1800-2005, 2009 and 2017):

> execute_region {
>   while (region is nonempty) {
>     E = any event from region;
>     remove E from the region;
>     if (E is an update event) {
>       update the modified object;
>       schedule evaluation event for any process sensitive to the object;
>     } else { /* E is an evaluation event */
>       evaluate the process associated with the event and possibly schedule further events for execution;
>     }
>   }
> }

Which indicates our current implementation could be correct. I found what you pasted in 1364-2005.

caryr, 2020-12-06:
It looks like quote removes the beginning spaces so here it is as code
```
execute_region {
    while (region is nonempty) {
        E = any event from region;
        remove E from the region;
        if (E is an update event) {
            update the modified object;
            schedule evaluation event for any process sensitive to the object;
        } else { /* E is an evaluation event */
            evaluate the process associated with the event and possibly schedule further events for execution;
        }
    }
}
```

martinwhitaker, 2020-12-06:
Please consider my comment of 10th May. It is the second example where I now think the current Icarus behaviour is incorrect, on the basis that if the always begin-end block is atomic because it has a timing control, the continuous assignment cannot be evaluated between the two procedural assignments, hence the wires can't glitch, even if the regs do.

caryr, 2020-12-06:
What is the rational for assuming the begin/end is atomic? Any reference would be appreciated. Yes, in a single threaded simulator it could appear atomic, but in a multi threaded environment you would have to lock the variable at the block level not the individual assignment statement level and then the update event code would have to filter the assignment glitch based on the value at locking versus unlocking the variable. I have not seen in the standard where that is specified or implied.

Also I don't think you can just filter the assignment at the process level since if you use the "flag" variable as an R-value between the two assignments the correct value needs to be used. This gets even more complicated when you consider multiple processes writing to or a different concurrent process reading the same variable. With some kind of lock you could at least detect the collision and report them as an issue.

I think what you want is conceptually acquire an exclusive lock for a variable whenever a process writes to it. The lock is held until the process reaches a stop point. It is at this point that any update events are created. Any other process that tries to read or write during the lock period should be reported as a concurrent access and terminate the simulation. You could use the previous value (value at locking) for a different process read and generate a warning message, but waiting for the unlock could lead to a dead lock. Reading from the local process would be okay since it has the lock on the variable. I think you would need this basic infrastructure even for single threaded operation to deal with bugs related to lock release. I think this could be added by extending the variable read/write routines to include their process information and then add an unlock at the stop point, but you likely need to assume any VPI access is from the local process so not 100% safe.

steveicarus, 2020-12-07:
The begin-end running to completion (or until it explicitly yields) is an unwritten rule that is a requirement to make any synchronization at all work. There is no language support for thread synchronization, so it had to be that sequential statements cannot be preempted.

caryr, 2020-12-07:
Yes, I hope nothing I wrote implied that was required. It's just what we "currently" do in Icarus.

I think the "Fine-grain process control" section of the standard should also be reviewed (1800-2012 section 9.7) and specifically the suspend()/resume() methods. How do you implement atomic behavior when a process can be suspended at any point in time and then later resumed? We often focus on the currently implemented behavior and the specifications around it, but the unimplemented sections often provide insight or restrictions on how things should be implemented (e.g. the VPI interface dictates items and access structure in the implementation and this not being considered consistently has puts significant limitations on what we can support from the VPI).

shlemiel, 2023-02-22:
I ran into the same problem and investigated a bit. The problem, I think, is that the style of initializing the FSM to 0 causes problems only in iverilog, when specific conditions are met, even though it is used relatively often in HDL designs. Let me elaborate a PoC of the hang in below:

```
module testbench;
    reg     clk;
    reg     rst_n;
    reg     start;
    wire    ready;
    exec_unit u_exec_unit(
        clk, rst_n, start, ready
    );

    initial begin
        clk = 0;
        
        forever begin
            #5 clk = ~clk;
        end
    end

    initial begin
        start <= 0;
        rst_n <= 0;
        @(posedge clk);
        @(posedge clk);
        rst_n <= 1;
        @(posedge clk);
        start <= 1;
        @(posedge clk);
        start <= 0;

        while(~ready) begin
            @(posedge clk);
        end

        @(posedge clk);

        $display("Done");
        $finish;
    end
endmodule

module exec_unit(
    input       clk,
    input       rst_n,
    input       start,
    output reg  ready
);
    localparam ST_IDLE  = 2'b01;
    localparam ST_RUN   = 2'b10;

    reg     [1:0]       c_state, n_state;

    always @(posedge clk or negedge rst_n) begin
        if(!rst_n) begin
            c_state <= ST_IDLE;
        end else begin
            c_state <= n_state;
        end
    end

    wire    sub_exec_unit_ready;
    reg     start_sub_exec_unit;
    always @(*) begin
        $display("always1, %x", c_state);
        n_state = c_state;
        ready = 0;
        start_sub_exec_unit = 0;

        case(c_state)
            ST_IDLE: begin
                ready = 1;

                if(start) begin
                    start_sub_exec_unit = 1;
                    n_state = ST_RUN;
                end
            end
            ST_RUN: begin
                if(sub_exec_unit_ready) begin
                    n_state = ST_IDLE;
                end
            end
        endcase
    end

    //*
    reg     start_sub_exec_unit_noglitch;
    always @(*) begin
        start_sub_exec_unit_noglitch <= start_sub_exec_unit;
    end
    //*/

    sub_exec_unit u_sub_exec_unit(
        .clk        (clk),
        .rst_n      (rst_n),
        .start      (start_sub_exec_unit),
        // .start      (start_sub_exec_unit_noglitch),
        .ready      (sub_exec_unit_ready)
    );
endmodule

module sub_exec_unit(
    input       clk,
    input       rst_n,
    input       start,
    output reg  ready
);
    localparam ST_IDLE  = 2'b01;
    localparam ST_RUN   = 2'b10;

    reg     [1:0]       c_state, n_state;

    always @(posedge clk or negedge rst_n) begin
        if(!rst_n) begin
            c_state <= ST_IDLE;
        end else begin
            c_state <= n_state;
        end
    end

    always @(*) begin
        $display("always2, %x", c_state);
        n_state = c_state;
        ready = 0;

        case(c_state)
            ST_IDLE: begin
                ready = 1;

                if(start) begin
                    n_state = ST_RUN;
                end
            end
            ST_RUN: begin
                n_state = ST_IDLE;
            end
        endcase
    end
endmodule
```

I tested above code in commercial simulators like vivado / vcs, and the simulation went to the end with no problem.
In addition, the module above is considered fully synthesizable in common FPGA/ASIC toolchain like vivado / design compiler / etc.

In my opinion, in the above case, assigning a variable multiple times in single always block should not be considered as a glitch. The coding style applied in the above code is pretty common.

shlemiel, 2023-02-22:
I ran into the same problem and investigated a bit. The problem, I think, is that the style of initializing the FSM to 0 causes problems only in iverilog, when specific conditions are met, even though it is used relatively often in HDL designs. Let me elaborate a PoC of the hang in below:

```
module testbench;
    reg     clk;
    reg     rst_n;
    reg     start;
    wire    ready;
    exec_unit u_exec_unit(
        clk, rst_n, start, ready
    );

    initial begin
        clk = 0;
        
        forever begin
            #5 clk = ~clk;
        end
    end

    initial begin
        start <= 0;
        rst_n <= 0;
        @(posedge clk);
        @(posedge clk);
        rst_n <= 1;
        @(posedge clk);
        start <= 1;
        @(posedge clk);
        start <= 0;

        while(~ready) begin
            @(posedge clk);
        end

        @(posedge clk);

        $display("Done");
        $finish;
    end
endmodule

module exec_unit(
    input       clk,
    input       rst_n,
    input       start,
    output reg  ready
);
    localparam ST_IDLE  = 2'b01;
    localparam ST_RUN   = 2'b10;

    reg     [1:0]       c_state, n_state;

    always @(posedge clk or negedge rst_n) begin
        if(!rst_n) begin
            c_state <= ST_IDLE;
        end else begin
            c_state <= n_state;
        end
    end

    wire    sub_exec_unit_ready;
    reg     start_sub_exec_unit;
    always @(*) begin
        $display("always1, %x", c_state);
        n_state = c_state;
        ready = 0;
        start_sub_exec_unit = 0;

        case(c_state)
            ST_IDLE: begin
                ready = 1;

                if(start) begin
                    start_sub_exec_unit = 1;
                    n_state = ST_RUN;
                end
            end
            ST_RUN: begin
                if(sub_exec_unit_ready) begin
                    n_state = ST_IDLE;
                end
            end
        endcase
    end

    //*
    reg     start_sub_exec_unit_noglitch;
    always @(*) begin
        start_sub_exec_unit_noglitch <= start_sub_exec_unit;
    end
    //*/

    sub_exec_unit u_sub_exec_unit(
        .clk        (clk),
        .rst_n      (rst_n),
        .start      (start_sub_exec_unit),
        // .start      (start_sub_exec_unit_noglitch),
        .ready      (sub_exec_unit_ready)
    );
endmodule

module sub_exec_unit(
    input       clk,
    input       rst_n,
    input       start,
    output reg  ready
);
    localparam ST_IDLE  = 2'b01;
    localparam ST_RUN   = 2'b10;

    reg     [1:0]       c_state, n_state;

    always @(posedge clk or negedge rst_n) begin
        if(!rst_n) begin
            c_state <= ST_IDLE;
        end else begin
            c_state <= n_state;
        end
    end

    always @(*) begin
        $display("always2, %x", c_state);
        n_state = c_state;
        ready = 0;

        case(c_state)
            ST_IDLE: begin
                ready = 1;

                if(start) begin
                    n_state = ST_RUN;
                end
            end
            ST_RUN: begin
                n_state = ST_IDLE;
            end
        endcase
    end
endmodule
```

I tested the above code in commercial simulators like vivado / vcs, and the simulation went to the end with no problem.
In addition, the module above is considered fully synthesizable in common FPGA/ASIC toolchain like vivado / design compiler / etc.

I'm sorry that I don't have much knowledge in verilog standards. However, this case should be considered as a bug. A coding style that assigns multiple times in a single always block is pretty common to avoid latches and it is fully synthesizable. Most of case it would cause no problem, but it could be a problem in some other cases, and when it be a problem, many people won't be able to figure out the reason and move to other simulators. I hope your understanding.

Thank you.

steveicarus, 2023-02-26:
Closing this issue since as I understand it it is actually valid standard behavior.

closed 2023-02-26 by steveicarus