steveicarus/iverilog #322: VPI: does not return reg name starting with an underscore

cmarqu, 2020-05-07:
We have a test module with a single-bit reg starting with an underscore (``reg _underscore_name;``):
https://github.com/cocotb/cocotb/blob/50c9cb6d3ce8c6d21eb29067e31b0775427a0301/tests/designs/sample_module/sample_module.sv#L138
However, Icarus 10.3 does not return this reg via VPI.

See also https://github.com/cocotb/cocotb/pull/1634.

The following is the output that ``VPI_TRACE`` gives for that module:
<details>
  <summary>VPI_TRACE output</summary>

```
vpi_handle(vpiSysTfCall, 0) -> 0x1c00ee0 ($dumpfile)
vpi_iterate(89, 0x1c00ee0) ->
vpi_get(vpiType, 0x1bff6b0) --> vpiConstant
vpi_get(vpiConstType, 0x1bff6b0) --> 6
vpi_free_object(0x1be3b90) --> 1
vpi_handle(vpiSysTfCall, 0) -> 0x1c00e80 ($dumpvars)
vpi_iterate(89, 0x1c00e80) ->
vpi_get(vpiType, 0x1c04620) --> vpiConstant
vpi_get(vpiConstType, 0x1c04620) --> 3
vpi_get(vpiType, 0x1bd3bc0) --> vpiModule
vpi_free_object(0x1be3b90) --> 1
vpi_iterate(32, (nil)) ->
vpi_get(vpiType, 0x1bd3bc0) --> vpiModule
vpi_get_str(vpiFullName, 0x1bd3bc0) --> sample_module
vpi_free_object(0x1b2b780) --> 1
vpi_get_str(vpiFullName, 0x1bd3bc0) --> sample_module
vpi_get(vpiType, 0x1bd3bc0) --> vpiModule
vpi_handle_by_name(sample_module.stream_in_data, (nil)) -->
vpi_iterate(32, (nil)) ->
vpi_get_str(vpiName, 0x1bd3bc0) --> sample_module
vpi_free_object(0x1c64100) --> 1
vpi_get_str(vpiName, 0x1bd3bc0) --> sample_module
vpi_get(vpiType, 0x1bb2920) --> vpiPort
vpi_get(vpiType, 0x1b797f0) --> vpiPort
vpi_get(vpiType, 0x1bb2950) --> vpiPort
vpi_get(vpiType, 0x1bb28f0) --> vpiPort
vpi_get(vpiType, 0x1bcef50) --> vpiPort
vpi_get(vpiType, 0x1be10b0) --> vpiPort
vpi_get(vpiType, 0x1be1060) --> vpiPort
vpi_get(vpiType, 0x1be1170) --> vpiPort
vpi_get(vpiType, 0x1be1210) --> vpiPort
vpi_get(vpiType, 0x1be29b0) --> vpiParameter
vpi_get_str(vpiName, 0x1be29b0) --> NUM_OF_MODULES
vpi_get(vpiType, 0x1be29b0) --> vpiParameter
vpi_get(vpiType, 0x1be29b0) --> vpiParameter
vpi_get(vpiType, 0x1be3c70) --> vpiNet
vpi_get_str(vpiName, 0x1be3c70) --> and_output
vpi_get(vpiType, 0x1be3c70) --> vpiNet
vpi_get(vpiType, 0x1be3c70) --> vpiNet
vpi_get(vpiType, 0x1ba03b0) --> vpiMemory
vpi_get_str(vpiName, 0x1ba03b0) --> array_0_to_3
vpi_get(vpiType, 0x1ba03b0) --> vpiMemory
vpi_iterate(30, 0x1ba03b0) ->
vpi_get_str(vpiName, 0x1b39010) --> array_0_to_3[0]
vpi_get_str(vpiName, 0x1b39028) --> array_0_to_3[1]
vpi_get_str(vpiName, 0x1b39040) --> array_0_to_3[2]
vpi_get_str(vpiName, 0x1b39058) --> array_0_to_3[3]
vpi_free_object(0x1b46fe0) --> 1
vpi_get(vpiType, 0x1b7b9f0) --> vpiMemory
vpi_get_str(vpiName, 0x1b7b9f0) --> array_2d
vpi_get(vpiType, 0x1b7b9f0) --> vpiMemory
vpi_iterate(30, 0x1b7b9f0) ->
vpi_get_str(vpiName, 0x1bd3830) --> array_2d[0]
vpi_get_str(vpiName, 0x1bd3848) --> array_2d[1]
vpi_get_str(vpiName, 0x1bd3860) --> array_2d[2]
vpi_get_str(vpiName, 0x1bd3878) --> array_2d[3]
vpi_get_str(vpiName, 0x1bd3890) --> array_2d[4]
vpi_get_str(vpiName, 0x1bd38a8) --> array_2d[5]
vpi_get_str(vpiName, 0x1bd38c0) --> array_2d[6]
[... 730 lines cut ...]
vpi_get(vpiType, 0x1be3fd0) --> vpiReg
vpi_get_str(vpiName, 0x1be3fd0) --> temp
vpi_get(vpiType, 0x1be3fd0) --> vpiReg
vpi_get(vpiType, 0x1be3fd0) --> vpiReg
vpi_get(vpiType, 0x1bc8a90) --> vpiGenScope
vpi_get_str(vpiName, 0x1bc8a90) --> genblk1[0]
vpi_get(vpiType, 0x1bc8a90) --> vpiGenScope
vpi_get(vpiType, 0x1bc8a90) --> vpiGenScope
vpi_get(vpiType, 0x1bff6f0) --> vpiGenScope
vpi_get_str(vpiName, 0x1bff6f0) --> genblk1[1]
vpi_get(vpiType, 0x1bff6f0) --> vpiGenScope
vpi_get(vpiType, 0x1bff6f0) --> vpiGenScope
vpi_get(vpiType, 0x1bff890) --> vpiGenScope
vpi_get_str(vpiName, 0x1bff890) --> genblk1[2]
vpi_get(vpiType, 0x1bff890) --> vpiGenScope
vpi_get(vpiType, 0x1bff890) --> vpiGenScope
vpi_get(vpiType, 0x1bffb40) --> vpiGenScope
vpi_get_str(vpiName, 0x1bffb40) --> genblk1[3]
vpi_get(vpiType, 0x1bffb40) --> vpiGenScope
vpi_get(vpiType, 0x1bffb40) --> vpiGenScope
vpi_handle_by_name: DONE
vpi_free_object(0x1c579e0) --> 1
```

</details>

caryr, 2020-05-08:
This is failing because Icarus drops variable during compilation that are never used and your variable is never used. This could be considered a bug in the context of the VPI exclusively using the variable, but that's not what this report is. If you assign the variable a value or use it in some way the compiler should pass it to the runtime and then it should be visible using the VPI. If this is not the case please let me know and I will create a real VPI test case and debug the issue. If you think the dropping of variables is wrong please create a new report for that.

cmarqu, 2020-05-08:
Thanks, it works indeed as soon as I do something with the variable.
Dropping variables seems to be a behavior that is different from other simulators that I tried, but I wouldn't claim it's wrong unless we could find a standard saying so.

closed 2020-05-08 by cmarqu

martinwhitaker, 2020-05-09:
Note that you can prevent the compiler from dropping unused variables by editing the vvp.conf file and deleting the line `functor:nodangle`.