steveicarus/iverilog #323: VPI: support for bit-selects (vpi_get_handle_by_index())

cmarqu, 2020-05-09:
It would be nice if bit-selects (``myvector[0]``) could be supported in VPI.
Right now, a message like ``Sorry: Icarus does not currently support vpi_get_handle_by_index() for vpiNet objects (stream_in_data).`` is printed. This is coming from https://github.com/steveicarus/iverilog/blob/40d2a49b90f6a977238f754d3427823917d688a4/vvp/vpi_signal.cc#L684-L688

A cocotb test for this is https://github.com/cocotb/cocotb/blob/master/tests/test_cases/test_discovery/test_discovery.py#L110

caryr, 2020-05-28:
FYI I have made progress on this. The basic handle by index for a net/reg is working and I'm working on the implementation of the bit functionality. The basic infrastructure for a bit type is in place and I am working on the get and put routines now. I will also need to add the iterators for the bit functionality, but I assume for cocotb you only need the get/put and handle by index so will push those changes once they are complete.

cmarqu, 2020-05-28:
Thanks very much!

caryr, 2020-05-31:
I have pushed support for vpi_by_index() of registers/nets to master. There is also support for getting the value from a bit select handle. I still need to work on the ability to put a value and bit iterators are not implemented. The put should print an appropriate message. Please verify that get from a bit handle works as you expect. I will work on put next followed by the iterators.

caryr, 2020-05-31:
Basic put value is now supported. This may be the last of the functionality you need for this report. Put value is still missing support for force/release and this may be complicated to add. I plan to look at that next. Also, unless scheduling is handled at a higher level, put value delays are not currently supported either.

caryr, 2020-06-05:
Have you been able to verify if the changes I have committed solve the issues in cocotb?

caryr, 2020-07-03:
Have you been able to check cocotb works with this change?

cmarqu, 2020-07-03:
Oops, sorry, I must have accidentally unsubscribed from this issue, so I didn't see your updates.
I'll try to check this over the weekend and let you know.

cmarqu, 2020-07-06:
Thanks very much for this, I can confirm that the two testcases now run fine.

closed 2020-07-06 by cmarqu