steveicarus/iverilog #324: cygwin build problem

staceycampbell, 2020-05-10:
On a clean new Cygwin64 installation on Windows 7:
```
$ git clone git@github.com:steveicarus/iverilog.git
$ cd iverilog
$ ./autogen.sh
$ ./configure
$ make
```
...later...

```
g++ -shared -Wl,--enable-auto-image-base -o system.vpi sys_table.o sys_convert.o sys_countdrivers.o sys_darray.o sys_deposit.o sys_display.o sys_fileio.o sys_finish.o sys_icarus.o sys_plusargs.o sys_queue.o sys_random.o sys_random_mti.o sys_readmem.o sys_readmem_lex.o sys_scanf.o sys_sdf.o sys_time.o sys_vcd.o sys_vcdoff.o vcd_priv.o mt19937int.o sys_priv.o sdf_parse.o sdf_lexor.o stringheap.o vams_simparam.o table_mod.o table_mod_parse.o table_mod_lexor.o sys_lxt.o lxt_write.o sys_lxt2.o lxt2_write.o sys_fst.o fstapi.o fastlz.o lz4.o vcd_priv2.o -L.  -lvpi -lbz2 -lz -lpthread -lhistory -lreadline 
/usr/lib/gcc/x86_64-pc-cygwin/9.3.0/../../../../x86_64-pc-cygwin/bin/ld: sys_table.o: in function `sys_lxt_or_vcd_register':
/cygdrive/c/Users/Stacey/src/iverilog/vpi/sys_table.c:83: undefined reference to `vpi_get_vlog_info'
/cygdrive/c/Users/Stacey/src/iverilog/vpi/sys_table.c:83:(.text+0x46): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `vpi_get_vlog_info'
/usr/lib/gcc/x86_64-pc-cygwin/9.3.0/../../../../x86_64-pc-cygwin/bin/ld: /cygdrive/c/Users/Stacey/src/iverilog/vpi/sys_table.c:192: undefined reference to `vpi_mcd_printf'
```
...many more undefined references follow.

martinwhitaker, 2020-05-10:
Fixed in both the master and v10 branches. The issue was that we still used __CYGWIN32__ in places, which is not defined by the Cygwin 64-bit compiler.

I note there are some test suite failures when using Cygwin. MSYS2 and the MinGW-w64 compiler are what we normally support and recommend.

closed 2020-05-10 by martinwhitaker

martinwhitaker, 2020-05-10:
Turns out the test suite failures were line-ending problems - solved by a clean clone of the test suite.