steveicarus/iverilog #326: Enable travis-ci

themperek, 2020-05-15:
Since there is already a `.travis.yml` in the repository it would beneficial just to enable it to run?

Howto: https://docs.travis-ci.com/user/tutorial/#to-get-started-with-travis-ci-using-github

martinwhitaker, 2020-06-01:
IIUC, only Steve can do this.

martinwhitaker, 2020-06-01:
But having just looked at a pull request, it looks like it is enabled...

themperek, 2020-06-01:
Looks not:  https://travis-ci.org/github/steveicarus

martinwhitaker, 2020-06-02:
Look here: https://travis-ci.com/github/steveicarus/iverilog/builds

With some effort, I've fixed the Windows builds. I've also added a check that the tests produce the expected results, and added the VPI tests.

What seems to be missing from this process is a notification that the CI fails after you push some changes.

closed 2020-06-02 by themperek