steveicarus/iverilog #330: ivl: eval_vec4.c:1091: draw_ternary_vec4: Assertion `tmp_flag >= 8' failed.

cdhmanning, 2020-05-30:
Hi Steve/All

I have a verilog file that is getting the above assertion with both 10.1 (what's in Ubuntu 18.04 ) and hot off git.

I'll have a crack at debugging/understanding the issue further.

Unfortunately the verilog code is not mine so I cannot supply the verilog, but if I will make an obfuscated version.

Thanks for a great tool!

Thanks

Charles

$ iverilog -v evil_file.v 
Icarus Verilog version 11.0 (devel) (s20150603-847-ga1518b57)

Copyright 1998-2019 Stephen Williams
<snip>
translate: /usr/local/lib/ivl/ivlpp  -v -L -F"/tmp/ivrlg27de7d9ac" -f"/tmp/ivrlg7de7d9ac" -p"/tmp/ivrli7de7d9ac" | /usr/local/lib/ivl/ivl -v -C"/tmp/ivrlh7de7d9ac" -C"/usr/local/lib/ivl/vvp.conf" -- -
Icarus Verilog Preprocessor version 11.0 (devel) (s20150603-847-ga1518b57)

Copyright (c) 1999-2017 Stephen Williams (steve@icarus.com)

<snip>
Using language generation: IEEE1364-2005,no-specify,xtypes,icarus-misc
PARSING INPUT
LOCATING TOP-LEVEL MODULES
  evil_file
 ... done, 0 seconds.
ELABORATING DESIGN
 ... done, 0 seconds.
RUNNING FUNCTORS
 -F cprop ...
 ... Iteration detected 0 optimizations.
 ... Look for dangling constants
 ... done
 -F nodangle ...
 ... scan for dangling signal and event nodes. (scomplete=F, ecomplete=F)
 ... 1 iterations deleted 2 dangling signals and 0 events.
 ... scan for dangling signal and event nodes. (scomplete=T, ecomplete=F)
 ... 2 iterations deleted 2 dangling signals and 0 events.
 ... done
CALCULATING ISLANDS
 ... done, 0 seconds.
CODE GENERATION
 ... invoking target_design
ivl: eval_vec4.c:1091: draw_ternary_vec4: Assertion `tmp_flag >= 8' failed.

cdhmanning, 2020-05-31:
Here is the naughty module that triggers this problem.

[naught_module.v.zip](https://github.com/steveicarus/iverilog/files/4706980/naught_module.v.zip)

Adding a printf reveals that the flag allocation fails due to trying to allocate too many flags.

$ iverilog naught_module.v
tmp_flag = 8
...
tmp_flag = 252
tmp_flag = 253
tmp_flag = 254
tmp_flag = 255
tmp_flag = -1
ivl: eval_vec4.c:1093: draw_ternary_vec4: Assertion `tmp_flag >= 8' failed.

The following patch makes the problem go away:
diff --git a/tgt-vvp/vvp.c b/tgt-vvp/vvp.c
index 9ab70ebf..d9441db2 100644
--- a/tgt-vvp/vvp.c
+++ b/tgt-vvp/vvp.c
@@ -50,7 +50,7 @@ unsigned show_file_line = 0;
 
 int debug_draw = 0;
 
-# define FLAGS_COUNT 256
+# define FLAGS_COUNT 512
 
 static uint32_t allocate_flag_mask[FLAGS_COUNT / 32] = { 0x000000ff, 0 };
 
@@ -251,3 +251,4 @@ const char* target_query(const char*key)
 
       return 0;
 }

Is that a realistic way to fix the issue?

caryr, 2020-05-31:
I will take a look at your suggested fix. I would prefer that this give an appropriate message as well as increasing the limit.

caryr, 2020-05-31:
I have pushed a fix to both devel and V10 that increases the flag count in the tgt.vvp compiler and in VVP to 512 along with adding an error message when the current flag count is used up instead of returning a bad value that asserts. Which was not very user friendly.

Please verify that this both compiles and runs. Your proposed fix would have fixed the compile issue, but there would have likely been runtime issues since it still only had 256 flag locations.

caryr, 2020-06-05:
Have you been able to verify this fully solves the issue with the original code?

cuijialang, 2020-06-30:
i find a bug when make in centos
[image]

martinwhitaker, 2020-06-30:
@cuijialang, you need to install all the necessary tools to build Icarus, as described in the README.txt file or on the Wiki (https://iverilog.fandom.com/wiki/Installation_Guide). From your screenshot, you haven't installed gperf.

If you have further problems, please open a new issue instead of hijacking this one.

caryr, 2020-07-03:
No response, so I will assume the committed change solved the issue.

closed 2020-07-03 by caryr