steveicarus/iverilog #333: Port Connection Error

codey-boi, 2020-06-11:
### Issue Summary
A port in my module "mod1" is not recognized as existing. It is declared correctly and works as intended in an alternate Verilog compiler.

### Small Reproducible Example (in file my_first.v)
```
module mod1 (a, d[2:0], d[3]);
    input a, b, c;
    output [3:0] d;
endmodule

module MY_MODULE (pin1, pin2);
    input pin1;
    output [2:0] pin2;
    mod1 I_mod1 (.a(pin1), .d( { pin5[2:0], pin4} ) );
endmodule
```

### Error
```
my_first.v:9: error: port ``d'' is not a port of I_mod1.
1 error(s) during elaboration.
```

Sahil1216, 2020-06-12:
[image]
Try this! Might work #333

martinwhitaker, 2020-06-12:
From IEEE 1364-2005 section 12.3.6:

"The second way to connect module ports consists of explicitly linking the two names for each side of the connection: the port declaration name from the module declaration to the expression, i.e., the name used in the module declaration, followed by the name used in the instantiating module. This compound name is then placed in the list of module connections. The port name shall be the name specified in the module declaration. The port name cannot be a bit-select, a part-select, or a concatenation of ports."

And earlier in section 12.2.3 there is this example:
````
module split_ports (a[7:4], a[3:0]);
// First port is upper 4 bits of 'a'.
// Second port is lower 4 bits of 'a'.
// Can't use named port connections because
// of part-select port 'a'.
````
So I believe your code is not valid Verilog.

codey-boi, 2020-06-12:
@martinwhitaker Thank you for this explanation, I must have completely missed that last sentence while looking through the manual.

closed 2020-06-12 by codey-boi