steveicarus/iverilog #338: iverilog won't build under OS X Catalina

dredozubov, 2020-06-30:
I've been trying to build this project with:

```
sh autoconf.sh
./configure 
CC="cc -no-cpp-precomp" ./configure
make
```

I'm getting:

```
g++ -I. -I.. -I../libmisc -I/usr/local/opt/icu4c/include -I/usr/local/opt/openssl/include -DHAVE_CONFIG_H -Wall -Wextra -Wshadow   -g -O2 -MD -c vtype_stream.cc -o vtype_stream.o
mv vtype_stream.d dep/vtype_stream.d
flex -s -olexor.cc ./lexor.lex
bison --verbose -t -d -o parse.cc parse.y
parse.y:2.9-16: syntax error, unexpected identifier, expecting string
make[1]: *** [parse.cc] Error 1
make: *** [all] Error 2
```

I've been trying to follow `macosx.txt`(it would be a good idea to mention it in README) and `dlcompat` doesn't seem buildable also:

```
[~/workspace/iverilog/dlcompat-20010831] % make
cc -Wall -O2 -DDEBUG=0 -fno-common -o dlopen.o -c dlopen.c
dlopen.c:71:58: warning: incompatible pointer to integer conversion initializing 'dev_t' (aka 'int') with an
      expression of type 'void *' [-Wint-conversion]
static const struct dlopen_handle main_program_handle = {NULL};
                                                         ^~~~
/Library/Developer/CommandLineTools/usr/lib/clang/11.0.0/include/stddef.h:105:16: note: expanded from macro
      'NULL'
#  define NULL ((void*)0)
               ^~~~~~~~~~
dlopen.c:81:6: error: expected identifier or '{'
enum bool
     ^
/Library/Developer/CommandLineTools/usr/lib/clang/11.0.0/include/stdbool.h:31:14: note: expanded from macro
      'bool'
#define bool _Bool
             ^
dlopen.c:515:2: error: expected identifier or '('
}
 ^
1 warning and 2 errors generated.
make: *** [dlopen.o] Error 1
```

martinwhitaker, 2020-06-30:
What version of bison are you using?

macos.txt was written in 2002. The instructions on the Wiki (https://iverilog.fandom.com/wiki/Installation_Guide#Compiling_on_Macintosh_OS_X) suggest that it is outdated.

dredozubov, 2020-06-30:
Thanks for pointing it out, using newer bison from brew allowed it to progress and build. Although after running `make install` and running the build executable I get `Cannot locate IVL modules : couldn't get command path from OS.`

martinwhitaker, 2020-06-30:
Force a rebuild of driver/main.c and check the compile command line. It should include a -DIVL_ROOT=... macro definition. What path is defined, and has that path been created when you ran 'make install'?

dredozubov, 2020-06-30:
Forcing a rebuild helped, thanks. `master` builds, so I guess the issue can be closed.

closed 2020-07-03 by caryr