steveicarus/iverilog #341: Wrong warning message for $fread

jotego, 2020-07-04:
$fread provides a warning when reading files past the end of the memory range. However, it seems to be off by one.

Try this:
```
    reg [7:0] foo [0:127];
    file=$fopen("foo.bin","rb");
    cnt=$fread(foo,file,0,64);
    cnt=$fread(foo,file,64,64);
    $fclose(file);
```
You will get this message:
```
WARNING: foo.v:68: $fread's count argument (64) is too large for start (64) and memory range [0:127].
```
But we are clearly only loading 128 bytes into a 128-long buffer so there should be no warning.

caryr, 2020-07-04:
This should now be fixed in the latest V10 and devel.

closed 2020-07-08 by caryr

jotego, 2020-12-15:
I still see this issue in version 12, and in the latest code as I just compiled and still got the error messages. Please, reopen the issue.