steveicarus/iverilog #345: Assertion failure while elaborating generate loop

zachjs, 2020-07-16:
iverilog appears to have trouble with constant function calls in certain contexts of generate constructs.

For example, on https://github.com/YosysHQ/yosys/blob/master/tests/various/const_func.v, I get:
```
Assertion failed: (test), function generate_scope_loop_, file elab_scope.cc, line 1172.
sh: line 1: 24775 Done                    /usr/local/Cellar/icarus-verilog/10.3/lib/ivl/ivlpp -L -F"/var/folders/15/qlmr890j0gq81stdzpt3t6xw0000gn/T//ivrlg218d12583" -f"/var/folders/15/qlmr890j0gq81stdzpt3t6xw0000gn/T//ivrlg18d12583" -p"/var/folders/15/qlmr890j0gq81stdzpt3t6xw0000gn/T//ivrli18d12583"
     24776 Abort trap: 6           | /usr/local/Cellar/icarus-verilog/10.3/lib/ivl/ivl -C"/var/folders/15/qlmr890j0gq81stdzpt3t6xw0000gn/T//ivrlh18d12583" -C"/usr/local/Cellar/icarus-verilog/10.3/lib/ivl/vvp.conf" -- -
```

I am using `Icarus Verilog version 10.3 (stable) (v10_3)`. If it would be helpful, I can produce other examples as well.

martinwhitaker, 2020-07-16:
This is fixed in the master branch (issue #146), but the fix was not back-ported to v10.

martinwhitaker, 2020-07-16:
I've back-ported the fix to the v10 branch.

closed 2020-07-16 by martinwhitaker