steveicarus/iverilog #348: New release?

jeanthom, 2020-07-29:
Hi!

Lots of changes have happened on gram since its last release. This means that most Linux distro packages are out of date. When can we expect a new release?

steveicarus, 2020-07-29:
> Lots of changes have happened in iverilog since its last release. This

Good question.

I'd say go to the iverilog-devel mailing list and ask that very question.
We may well decide that it is time for version 11.

caryr, 2020-07-30:
There are things I'm working on now as well as a few things I need to finish up or start that I would like to be complete before we do an 11 release. Though I agree there are certainly a lot of changes we would like to get into the distributions.

Specifically I want to get the longest static prefix working for the always_* block sensitivity calculation. Finish the queue work I have been working on for the last few weeks and then some other thing on my want list.

In case you haven't noticed, lots of changes have been pushed into development over the last few weeks.

martinwhitaker, 2020-07-30:
I think a v11 release is long overdue - there are a lot of bug fixes that haven't and can't be back-ported to v10. Let's get something out sooner rather than later.

steveicarus, 2020-07-30:
Let's move this conversation to the iverilog-devel mailing list.

martinwhitaker, 2020-10-12:
Steve has released v11.

closed 2020-10-12 by martinwhitaker