steveicarus/iverilog #350: Add rich rule checking

cuijialang, 2020-08-03:
Is your feature request related to a problem? Please describe.
Hello, do you have plans to add richer rules to check for syntax errors, such as Leda / spyglass, etc
Is there a similar set of policy，such as
RMM，DO-254，ESSENTIAL..
Describe the solution you'd like
Can you direct us on how to add this feature to your version or modify those files / source code.

Describe alternatives you've considered
I want to add these rules, but I am still a novice and need your guidance and help

martinwhitaker, 2020-08-04:
Icarus already has strict checks for syntax errors. I guess what you are asking for is a linting tool. There used to be a third-party tool based on v0.8/v0.9 (described at https://iverilog.fandom.com/wiki/User_Guide#Add-on_Products), but the link is dead.

I don't think any of the Icarus developers has plans to add such a feature.

cuijialang, 2020-08-05:
yes,you are right
Can you tell me how to add these rule sets,As a plug-in
Checklist_ruleset、DO-254_ruleset、RMM_rulesetSafety-Critical_ruleset、Verification_ruleset、Essentials_ruleset、RTLC_ruleset

cuijialang, 2020-08-05:
[RMM_ruleset.zip](https://github.com/steveicarus/iverilog/files/5025529/RMM_ruleset.zip)
this is the rulesets

xinlnix, 2020-08-05:
Iverilog don't support always@(*) include reg  statement error check,such as:
`
always @(posedge clk)
Q <= A & (B | C);
always @ (E)
Q = E;
`
Somebody who can tell me how to change the source code to check this statement error?
Thanks a lot.

martinwhitaker, 2020-08-05:
@xinlnix, that is perfectly legal Verilog, so a simulator won't report an error. However, if you enable the iverilog synthesis option (-S), it will report
````
test.v:7: warning: 'Q' is driven by more than one process.
test.v:7: sorry: Cannot synthesize signals that are driven by more than one process.
````
If you study the compiler code to see what caused those messages, that may give you some ideas on how to do what you want.

xinlnix, 2020-08-05:
@martinwhitaker But 0.9 or later iverilog don't support -S option
`-S (v0.8 only) Edit
Activate synthesis. This flag tells the compiler to do what synthesis it can do before calling the code generator. This flag is rarely used explicitly, and certain code generators will implicitly enable this flag.

NOTE: Icarus Verilog release 0.9 and later do not support synthesis, even though this flag is still present. Please do not expect this flag to do anything useful for you in 0.9 or later releases.`
How to deal with it?

xinlnix, 2020-08-05:
@martinwhitaker from this [page](https://iverilog.fandom.com/wiki/Iverilog_Flags)

martinwhitaker, 2020-08-05:
The Wiki was out of date. The -S option was reenabled in v10.

xinlnix, 2020-08-06:
My iverilog version is 10.1(stable)，but when I type`iverilog -S conflict.v ` ,the command line report nothing. conflict.v is
`
module conflict(
    input clk,
    input a,
    input b,
    input e,
    output  reg q
);
    always@(posedge clk) begin
        q <= a&b;
    end
always@(*) begin
       q = e;
   end
endmodule
`
The only output file is a.out(vvp file).

xinlnix, 2020-08-06:
@martinwhitaker Should I upgrade to version 10.3?

xinlnix, 2020-08-06:
@martinwhitaker thank u very much.I upgrade to 11(devel)version it show the error.
thanks again.

closed 2020-08-06 by cuijialang