steveicarus/iverilog #353: Segfault on key UP after a command is in history

hinxx, 2020-08-04:
This is running today's GIT code.
I get a segfault on key UP after a (single) command is in history. I've used the counter.v and counter_tb.v code from the https://iverilog.fandom.com/wiki/Getting_Started page.

I can issue key UP *before* entering any commands in the interactive prompt and I do **not** get a segfault, as soon as a single command is in history, pressing key UP results in a segfault. This is on Linux Mint 19.3, using libreadline6 so it seems.

```
hinxx@obzen ~/Projects/ice40/tmp $ ../iverilog/driver/iverilog counter*.v -o counter
hinxx@obzen ~/Projects/ice40/tmp $ gdb ../iverilog/vvp/vvp 
GNU gdb (Ubuntu 8.1-0ubuntu3.2) 8.1.0.20180409-git
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ../iverilog/vvp/vvp...done.
(gdb) r counter
Starting program: /home/hinxx/Projects/ice40/iverilog/vvp/vvp counter
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
At time                    0, value = xx (x)
At time                   17, value = 00 (0)
At time                   35, value = 01 (1)
At time                   45, value = 02 (2)
At time                   55, value = 03 (3)
At time                   57, value = 00 (0)
At time                   75, value = 01 (1)
At time                   85, value = 02 (2)
At time                   95, value = 03 (3)
At time                  105, value = 04 (4)
At time                  115, value = 05 (5)
At time                  125, value = 06 (6)
At time                  135, value = 07 (7)
At time                  145, value = 08 (8)
At time                  155, value = 09 (9)
At time                  165, value = 0a (10)
** VVP Stop(0) **
** Flushing output streams.
** Current simulation time is 168 ticks.
> 
> 
> help
Commands can be from the following table of base commands,
or can be invocations of system tasks/functions.

cd       - Synonym for push.
cont     - Resume (continue) the simulation
finish   - Finish the simulation.
help     - Get help.
list     - List items in the current scope.
load     - Load a VPI module, a la vvp -m.
ls       - Shorthand for "list".
pop      - Pop one scope from the scope stack.
push     - Descend into the named scope.
step     - Single-step the scheduler for 1 event.
time     - Print the current simulation time.
trace    - Control statement tracing (on/off) when the code is instrumented.
where    - Show current scope, and scope hierarchy stack.

If the command name starts with a '$' character, it
is taken to be the name of a system task, and a call is
built up and executed.
> 
```

At this point I've pressed key UP.

```
Program received signal SIGSEGV, Segmentation fault.
0x00007ffff7795a0b in previous_history () from /lib/x86_64-linux-gnu/libreadline.so.6
(gdb) bt
#0  0x00007ffff7795a0b in previous_history () from /lib/x86_64-linux-gnu/libreadline.so.6
#1  0x00007ffff77955e5 in rl_get_previous_history () from /lib/x86_64-linux-gnu/libreadline.so.6
#2  0x00007ffff777a990 in _rl_dispatch_subseq () from /lib/x86_64-linux-gnu/libreadline.so.6
#3  0x00007ffff777ab14 in _rl_dispatch_subseq () from /lib/x86_64-linux-gnu/libreadline.so.6
#4  0x00007ffff777ab14 in _rl_dispatch_subseq () from /lib/x86_64-linux-gnu/libreadline.so.6
#5  0x00007ffff777ae12 in readline_internal_char () from /lib/x86_64-linux-gnu/libreadline.so.6
#6  0x00007ffff777b545 in readline () from /lib/x86_64-linux-gnu/libreadline.so.6
#7  0x00005555555ed5b8 in stop_handler (rc=rc@entry=0) at stop.cc:552
#8  0x0000555555609fee in schedule_simulate () at schedule.cc:1156
#9  0x00005555555bf1bd in main (argc=<optimized out>, argv=0x7fffffffddc8) at main.cc:465
(gdb) 
```

hinxx, 2020-08-04:
FYI, I'm seeing the same issue if I try the similar interaction using a small piece of code using libreadline and compiling with `gcc  -O0 -ggdb3 readline.c -o readline -lhistory -lreadline`.

The issue does not happen I the compilation does not include `-lhistory`. Then the key UP works as expected (in both, my test code and `vvp`).

It seems that the issue is not specific to iverilog code, but my system. Sorry for the noise.

martinwhitaker, 2020-08-05:
I can reproduce this using Ubuntu 18.04, but not on my usual Linux system. Both have libreadline.so.7.

I think the problem is that both libreadline and libhistory export an add_history() function. By putting libhistory first on the linker command line, the function from libhistory gets used. I don't know why this should matter on some systems but not on others.

It's not clear why vvp is linked against both libhistory and libreadline. Maybe that was needed in the past. Perhaps Steve can remember.

hinxx, 2020-08-05:
I've also built my own libreadline7 for a test, and voided using system provided one. The problem went away. I think I rebuilt the iverilog vvp with the same linker line including -lhistory and were able to not get a segfault. 

That made me open up an Ubuntu bug report just to be told to ask on Mint forum instead. Waiting for more info from Mint folks.

OTOH, if linking in libhistory is redundant it might be interesting to know what is the reason for its use in vvp.. might be historical.

martinwhitaker, 2020-08-06:
I've modified the configure script to only use libhistory if the history functions are not available in libreadline.

martinwhitaker, 2020-08-07:
Could you test if this fixes/works around the bug for you (not sure if this is really a bug in libreadline or a bug in the way we are using it).

hinxx, 2020-08-07:
No more segfaults after pulling in new changes, rebuilding && reinstalling the tools.

FYI, I got the answer on the Linux Mint forum as well. It points to https://lists.gnu.org/archive/html/bug-readline/2015-10/msg00001.html message.

Here is the message for completeness sake:
```
> If I compile my program with:
>   g++ ... -lhistory -lreadline
> I get the problem.
> 
> While if I compile the program with the opposite order:
>   g++ ... -lreadline -lhistory
> Everything works.

If you're linking with libreadline, you shouldn't link with libhistory at
all -- the readline library incorporates the history library.  If you do,
though, you should make sure that the readline library comes first, as you
discovered.
```
Seems you made the right call in removing the `-lhistory` in favor of `-lreadline`.

martinwhitaker, 2020-08-08:
Thanks for testing. Closing as fixed.

closed 2020-08-08 by martinwhitaker