steveicarus/iverilog #356: tri0 vs pulldown inconsistency & conflicts

Johnlon, 2020-08-06:
I am running latest dev build built locally.

I wasn't able to extract a simple demo of the issue, however, my homebrew cpu simulation demonstrates the issue.
See  https://github.com/Johnlon/spam-1/commit/800672510378abe436cc41c4683532a86a5a218d
This commit is my master head at the time of writing.

**Steps to reproduce good behaviour using pulldown:**
install iverilog and vvp on the path
git clone the above
cd to the verilog subdirectory
cd to the cpu subdirectory
run ../simulate.sh from the cpu dir

Expect the test to run - it will spin in a loop running it's program.

**Steps to reproduce broken behaviour:**
Now edit the file cpu.v and change lines 48 and 49 from 
 48     tri [15:0] address_bus;
 49     pulldown pd1[15:0](address_bus);
to 
 48     tri0 [15:0] address_bus;
 49     //pulldown pd1[15:0](address_bus);

Rerun ../simulate.sh and it will now fail with a failed assertion ...

   8670ns  Line:216   FAILED: actual 'xxxxxxxxx0x0x0x0' != '1111111110101010' expected,   (dX!=d65450)(hxxXX!=hffaa)  : CPU.address_bus != 16'hffaa

This error is not expected.
The message appears to be suggesting the tri0 is conflicting with the bits that I expect to be set to 1 - this appears a bug.
The same doesn't happen with pulldown.

martinwhitaker, 2020-08-06:
Fixed in both master and v10 branches.

closed 2020-08-06 by martinwhitaker

Johnlon, 2020-08-06:
Just taken master and the bug persists.

Johnlon, 2020-08-06:
Hi - not fixed - pulled master just now.

martinwhitaker, 2020-08-06:
It's working here, following the procedure in your initial comment.

Johnlon, 2020-08-07:
Hmmm - yes I got ti to work thanks.

But ...

I've always been running it from the build directory using an absolute path
as I didn't want to install it.
However, even having rebuit it the change wasn't working.

It was only after doing an "make install" that it worked.

I certainly had an older version installed in /usr but I don't understand
why it might have needed overwriting.

I did an ldd to see any links and didn't see anything that would explain
why running with the abs path wouldn't work.
I assume the program reaches into /usr or something?

$ ldd /home/john/wslapps/iverilog/iverilog//vvp/vvp
        linux-vdso.so.1 (0x00007fffd7341000)
        libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0
(0x00007f60843e0000)
        libhistory.so.7 => /lib/x86_64-linux-gnu/libhistory.so.7
(0x00007f60841d0000)
        libreadline.so.7 => /lib/x86_64-linux-gnu/libreadline.so.7
(0x00007f6083f80000)
        libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f6083d70000)
        libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6
(0x00007f60839e0000)
        libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f6083640000)
        libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1
(0x00007f6083410000)
        libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f6083010000)
        /lib64/ld-linux-x86-64.so.2 (0x00007f6084c00000)
        libtinfo.so.5 => /lib/x86_64-linux-gnu/libtinfo.so.5
(0x00007f6082de0000)

$ ldd /home/john/wslapps/iverilog/iverilog/driver/iverilog
        linux-vdso.so.1 (0x00007fffd4c12000)
        libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007fdba2780000)
        /lib64/ld-linux-x86-64.so.2 (0x00007fdba2e00000)

martinwhitaker, 2020-08-08:
By default the iverilog driver looks for its sub-programs and config files in the install directory. You can override that using the -B option, as described in the man page.