steveicarus/iverilog #358: wire type can't be assignment by always@(*) block?

xinlnix, 2020-08-08:
```verilog
module conflict(
    input clk,
    input a,
    input b,
    input e,
    output  reg [3:0] q
);
    integer i;
    wire [3:0] wire_q;

    always@(*) begin
        wire_q[i] = e&b ;
   end
    always@(posedge clk) begin
        for(i=0;i<4;i++) begin
            q[i] <= a&b ;
        end
    end
endmodule
```
my code is above. The error messege is 
```
conflict.v:13: error: wire_q[i] is not a valid l-value in conflict.
conflict.v:10:      : wire_q[i] is declared here as wire.
Elaboration failed
```
**my version is:**
`Icarus Verilog Preprocessor version 11.0 (devel) (s20150603-911-gd1d6c0f5-dirty)`

jeanthom, 2020-08-08:
Your verilog code is invalid: you cannot write to wires inside an always block. Use assign instead.

xinlnix, 2020-08-08:
@jeanthom thanks a lot

closed 2020-08-08 by xinlnix