steveicarus/iverilog #360: Where can I find pform file format?

xinlnix, 2020-08-13:
where to find pform(parsed form) file format except pform.h?

martinwhitaker, 2020-08-14:
Normally the pform is not written to a file - it is just a set of data structures in memory that hold the results of parsing the input files. The data structure definitions are spread across a number of header files - 'ls P*.h' will find most of them. The pform can be written out in human-readable form for debugging purposes - see the README.txt file for instructions. The functions that do that can be found in pform_dump.cc.

All input files have to be parsed before you can elaborate a design and start building the netlist.

steveicarus, 2020-08-14:
If you are asking specifically about the pform file that it output is certain modes, that is a debugging aid (for debugging the compiler) and not really useful outside that purpose. You don't care about pform files unless you are a developer working on the Icarus Verilog compiler.

xinlnix, 2020-08-17:
I really want to develop iverilog.So I want to find the pform file format define and netlist file format define.Thanks very much.

closed 2020-08-17 by xinlnix