steveicarus/iverilog #363: infinite loop if always_comb is used

ercanal, 2020-08-15:
I am not sure, this is a problem or not but my simulation is getting stuck in one tick point. So I converted all ```always_comb``` to ```always@*``` and tried again and simulation worked fine.

I compared .vvp files and noticed that 

While vvp file with ```always_comb``` has following lines
```
E_0x55e8f4bdad40 .event edge, v0x55e8f50ed9d0_0, v0x55e8f50ea450_0, v0x55e8f50ed9d0_0, v0x55e8f50ea450_0;
....
    %file_line 12 112 "Event wait (@) statement.";
Ewait_7 .event/or E_0x55e8f4bdad40, E_0x0;
    %wait Ewait_7;

```
vvp file with ```always@*```  has following lines
````
E_0x55f64ad4ad40 .event edge, v0x55f64b25e5c0_0, v0x55f64b25cd40_0;
.....
    %file_line 12 112 "Event wait (@) statement.";
    %wait E_0x55f64ad4ad40;
````

martinwhitaker, 2020-08-15:
One difference is because an always_comb is unconditionally triggered at time 0 (handled by the .event/or statement). Without seeing your code, I can't say any more.

ercanal, 2020-08-15:
Thanks for info.
I think my case is similar to #321. With various Ctrl-C  and continue steps,  the simulation continuing correctly in always_comb case.

ercanal, 2020-08-16:
Sorry  but I have one more question:
In line below, the parameter are looks like senstivity list but same signal is listed twice for always_comb case. Can this cause any loop?

```
E_0x55e8f4bdad40 .event edge, v0x55e8f50ed9d0_0, v0x55e8f50ea450_0, v0x55e8f50ed9d0_0, v0x55e8f50ea450_0;
```

martinwhitaker, 2020-08-16:
I don't see that in a simple test case. Can you show the code that caused it.

caryr, 2020-08-16:
Yes, please provide some example code. Just providing the always_comb block and the declarations for the variables in the block to produce the output above is better than nothing. It is odd that there are repeated items, but I doubt that is what is causing the issue.

ercanal, 2020-08-21:
I compiled the only one verilog file but couldn't get double sensitivity list only happens when trying to compile the testbench with all files. I can't my share my all design code so I need some time to generate the same error.

steveicarus, 2020-12-30:
Since this cannot be reproduced, I'm closing this.

closed 2020-12-30 by steveicarus